SpinalHDL/VexiiRiscv #1: Roadmap

Dolu1990, 2023-11-14:
- [x] Plugin API
- [x] Pipeline API
- [ ] Basic frontend
- [ ] Decoder
- [ ] Dispatcher
- [ ] Execute
- [ ] Load / Store

andreasWallner, 2023-12-18:
just to avoid duplicate work: iterative shift is on it's way, just like iterative mul

bitpasta, 2024-01-01:
just to let you know - i'm working on a radix2 divider here:
https://github.com/bitpasta/VexiiRiscv/tree/dev_divradix2

Dolu1990, 2024-01-11:
lolololol

```
[Progress] Start VexiiRiscv test simulation with seed 2

OpenSBI v0.8
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name       : NaxRiscv
Platform Features   : timer,mfdeleg
Platform HART Count : 1
Boot HART ID        : 0
Boot HART ISA       : rv32imasu
BOOT HART Features  : scounteren,mcounteren
BOOT HART PMP Count : 0
Firmware Base       : 0x80000000
Firmware Size       : 64 KB
Runtime SBI Version : 0.2

MIDELEG : 0x00000222
MEDELEG : 0x0000b109
[    0.000000] Linux version 5.10.1 (rawrr@rawrr) (riscv32-buildroot-linux-gnu-gcc.br_real (Buildroot 2020.11-rc3-8-g9ef54b7d0b) 10.2.0, GNU ld (GNU Binutils) 2.34) #2 SMP Wed Jan 26 14:18:17 CET 2022
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] Initial ramdisk at: 0x(ptrval) (8388608 bytes)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080400000-0x000000008fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080400000-0x000000008fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080400000-0x000000008fffffff]
[    0.000000] SBI specification v0.2 detected
[    0.000000] SBI implementation ID=0x1 Version=0x8
[    0.000000] SBI v0.2 TIME extension detected
[    0.000000] SBI v0.2 IPI extension detected
[    0.000000] SBI v0.2 RFENCE extension detected
[    0.000000] SBI v0.2 HSM extension detected
[    0.000000] riscv: ISA extensions aim
[    0.000000] riscv: ELF capabilities aim
[    0.000000] percpu: Embedded 10 pages/cpu s18700 r0 d22260 u40960
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64008
[    0.000000] Kernel command line: rootwait console=hvc0 earlycon=sbi root=/dev/ram0 init=/sbin/init
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Sorting __ex_table...
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 241280K/258048K available (4717K kernel code, 553K rwdata, 632K rodata, 166K init, 213K bss, 16768K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] riscv-intc: 32 local interrupts mapped
[    0.000000] random: get_random_bytes called from start_kernel+0x35c/0x4dc with crng_init=0
[    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x171024e7e0, max_idle_ns: 440795205315 ns
[    0.000053] sched_clock: 64 bits at 100MHz, resolution 10ns, wraps every 4398046511100ns
[    0.001054] Console: colour dummy device 80x25
[    0.001405] printk: console [hvc0] enabled
[    0.001405] printk: console [hvc0] enabled
[    0.001949] printk: bootconsole [sbi0] disabled
[    0.001949] printk: bootconsole [sbi0] disabled
[... 55 lines cut ...]
mmu		: sv32

root@buildroot:~# echo 1+2+3*4 | bc
echo 1+2+3*4 | bc
15
root@buildroot:~# micropython
micropython
MicroPython v1.13 on 2022-01-26; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import math

>>> math.sin(math.pi/4)

0.7071067811865475
>>> from sys import exit

>>> exit()

root@buildroot:~# ls /
ls /
bin      init     linuxrc  opt      run      tmp
dev      lib      media    proc     sbin     usr
etc      lib32    mnt      root     sys      var
root@buildroot:~# 
```

bitpasta, 2024-01-11:
Wow nice! Congratulations!

Dolu1990, 2024-04-08:
Brawww 

```shell
root@buildroot:~# cat /proc/cpuinfo 
processor	: 0
hart		: 0
isa		: rv32ima
mmu		: sv32

processor	: 1
hart		: 1
isa		: rv32ima
mmu		: sv32
#################
chocolate-doom -1 -timedemo demo1.lmp &
...
timed 5026 gametics in 2724 realtics (64.577827 fps)
#################
chocolate-doom -1 -timedemo demo1.lmp &
chocolate-doom -1 -timedemo demo1.lmp &
...
timed 5026 gametics in 2897 realtics (60.721436 fps)
timed 5026 gametics in 2918 realtics (60.284443 fps)
```

Brawwwwwww

bitpasta, 2024-04-09:
Nice! :) How does that compare to Vex and Nax?

Dolu1990, 2024-04-09:
It compare quite well also, not all option are turned on.

```
Vex   : timed 5026 gametics in 4866 realtics (36.150841 fps) ( no l2)
Vexii : timed 5026 gametics in 2724 realtics (64.577827 fps) (128KB-l2)
Nax  : timed 5026 gametics in 2375 realtics (74.067368 fps)  (128KB-l2)
```

Just tried now quad core and octo core works aswell.

Dolu1990, 2024-04-30:
```
root@nexys:~# cat /etc/*-release
PRETTY_NAME="Debian GNU/Linux trixie/sid"
NAME="Debian GNU/Linux"
VERSION_CODENAME=trixie
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@nexys:~# cat /proc/cpuinfo 
processor	: 0
hart		: 0
isa		: rv64imafdc
mmu		: sv39
mvendorid	: 0x0
marchid		: 0x5
mimpid		: 0x0

processor	: 1
hart		: 1
isa		: rv64imafdc
mmu		: sv39
mvendorid	: 0x0
marchid		: 0x5
mimpid		: 0x0

root@nexys:~# 
```

Dolu1990, 2024-05-03:
```
root@nexys:~# neofetch 
       _,met$$$$$gg.          root@nexys 
    ,g$$$$$$$$$$$$$$$P.       ---------- 
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux trixie/sid riscv64 
 ,$$P'              `$$$.     Kernel: 6.1.0-rc2+ 
',$$P       ,ggs.     `$$b:   Uptime: 17 hours, 47 mins 
`d$$'     ,$P"'   .    $$$    Packages: 1698 (dpkg) 
 $$P      d$'     ,    $$P    Shell: bash 5.2.15 
 $$:      $$.   -    ,d$$'    Resolution: 800x600 
 $$;      Y$b._   _,d$P'      WM: wmaker 
 Y$$.    `.`"Y$$$$P"'         Theme: Adwaita [GTK3] 
 `$$b      "-.__              Icons: Adwaita [GTK3] 
  `Y$$                        Terminal: /dev/pts/2 
   `Y$$.                      CPU: (2) 
     `$$b.                    Memory: 97MiB / 472MiB 
       `Y$$b.
          `"Y$b._                                     
              `"""                                    
```

jahagirdar, 2024-09-06:
Is CFU on the roadmap?

Dolu1990, 2024-09-06:
Hi,

Yes it is, a few people need it :)
Also, it shouldn't be a dififcult one to add.

maass-hamburg, 2025-11-05:
How about Zfinx and Zdinx? Basically f and d, but it uses the integer registers instead of separate floating point registers. Generally needs less resources, on the fpga and also in running code, as the floating point registers don't have to saved for interrupts and context switches

Dolu1990, 2025-11-10:
Hi @maass-hamburg 

Fondamentaly, it would not be too complicated to supports. Mostly reconfiguring / rewireing things differently depending the config.
But currently i do not have any plan to implement it.

Dolu1990, 2025-12-01:
@inochisa Nice ^^

> I need to achieve minimum change for non-H build.

Yes, very important to keep simpler configs not impacted by more hardware :)