SpinalHDL/VexiiRiscv #10: JNI error

nachiket, 2024-03-14:
I'm sure its something buggy in my setup, but I get this error:
```
[info] ../verilator/VVexiiRiscv__spinalWrapper.cpp:5:10: fatal error: jni.h: No such file or directory
[info]  #include <jni.h>
```

I do have a JDK installed, but do I need a specific version? or any other missing packages beyond the JDK?

Dolu1990, 2024-03-15:
JDK should be enough 
Maybe some platform / version issue
What JDK do you have installed ?
Are you on windows / linux / ... ?

nachiket, 2024-03-15:
```
java -version
openjdk version "11.0.19" 2023-04-18
OpenJDK Runtime Environment (build 11.0.19+7-post-Ubuntu-0ubuntu118.04.1)
OpenJDK 64-Bit Server VM (build 11.0.19+7-post-Ubuntu-0ubuntu118.04.1, mixed mode, sharing)
```

I'm on Ubuntu 18.04

nachiket, 2024-03-15:
I see a jni.h here
/usr/lib/jvm/java-8-openjdk-amd64/include/jni.h

nachiket, 2024-03-15:
javac -version
javac 1.8.0_362

I assume that's too old?

Dolu1990, 2024-03-15:
hmm probably yes, i'm on 

javac --version
javac 11.0.20.1

nachiket, 2024-03-15:
Ok upgrading the JDK fixed this issue.

But I'm now seeing a different error:
```
[error] Exception in thread "main" spinal.sim.SimFailure: Vexii hasn't commited anything for too long, last uop id 0x20
[error] 	at spinal.core.sim.package$.simFailure(package.scala:171)
[error] 	at vexiiriscv.test.VexiiRiscvProbe.$anonfun$checkCommits$1(VexiiRiscvProbe.scala:521)
[error] 	at vexiiriscv.test.VexiiRiscvProbe.$anonfun$checkCommits$1$adapted(VexiiRiscvProbe.scala:515)
[error] 	at scala.collection.IndexedSeqOptimized.foreach(IndexedSeqOptimized.scala:36)
[error] 	at scala.collection.IndexedSeqOptimized.foreach$(IndexedSeqOptimized.scala:33)
[error] 	at scala.collection.mutable.ArrayOps$ofRef.foreach(ArrayOps.scala:198)
[error] 	at vexiiriscv.test.VexiiRiscvProbe.checkCommits(VexiiRiscvProbe.scala:515)
[error] 	at vexiiriscv.test.VexiiRiscvProbe.$anonfun$new$2(VexiiRiscvProbe.scala:582)
[error] 	at spinal.core.sim.package$SimClockDomainPimper.$anonfun$onSamplings$2(package.scala:987)
[error] 	at spinal.core.sim.package$SimClockDomainPimper.$anonfun$onSamplings$2$adapted(package.scala:987)
[error] 	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
[error] 	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
[error] 	at spinal.core.sim.package$SimClockDomainPimper.$anonfun$onSamplings$1(package.scala:987)
[error] 	at spinal.core.sim.package$$anon$1.update(package.scala:205)
[error] 	at spinal.sim.SimManager.runWhile(SimManager.scala:324)
[error] 	at spinal.sim.SimManager.runAll(SimManager.scala:246)
[error] 	at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:608
[error] 	at spinal.core.sim.SimCompiled.doSimUntilVoid(SimBootstraps.scala:581)
[error] 	at vexiiriscv.tester.TestOptions.test(TestBench.scala:165)
[error] 	at vexiiriscv.tester.TestBench$.doIt(TestBench.scala:553)
[error] 	at vexiiriscv.tester.TestBench$.delayedEndpoint$vexiiriscv$tester$TestBench$1(TestBench.scala:481)
[error] 	at vexiiriscv.tester.TestBench$delayedInit$body.apply(TestBench.scala:480)
[error] 	at scala.Function0.apply$mcV$sp(Function0.scala:39)
[error] 	at scala.Function0.apply$mcV$sp$(Function0.scala:39)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
[error] 	at scala.App.$anonfun$main$1$adapted(App.scala:80)
[error] 	at scala.collection.immutable.List.foreach(List.scala:431)
[error] 	at scala.App.main(App.scala:80)
[error] 	at scala.App.main$(App.scala:78)
[error] 	at vexiiriscv.tester.TestBench$.main(TestBench.scala:480)
[error] 	at vexiiriscv.tester.TestBench.main(TestBench.scala)
[error] Nonzero exit code returned from runner: 1
[error] (Test / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 12 s, completed Mar 15, 2024, 10:34:05 AM
```

Dolu1990, 2024-03-15:
It mean that the CPU didn't commited anything since too long
Did you loaded some elf/bin ? if not that's normal

nachiket, 2024-03-15:
I’m just running the command from the guide

https://spinalhdl.github.io/VexiiRiscv-RTD/master/VexiiRiscv/HowToUse/index.html

`vexiiriscv.tester.TestBench --load-elf ext/NaxSoftware/baremetal/dhrystone/build/rv32ima/dhrystone.elf --trace-all`

Dolu1990, 2024-03-15:
Ahhh right, mul/div aren't anymore included in the default configuration, you need to add : 
--with-mul --with-div 

Also, here is an example of additional arguements you can use to improve IPC :

--with-btb --with-gshare --with-ras --decoders 2 --lanes 2 --with-aligner-buffer --with-dispatcher-buffer --with-late-alu --regfile-async --allow-bypass-from 0 --div-radix 4

closed 2025-02-12 by Dolu1990