SpinalHDL/VexiiRiscv #12: Vexiiriscv instructions

ztachip, 2024-04-06:
Could you provide some instructions on how to install and use this new version.
Is it similar to VexRiscv?
Thx

Dolu1990, 2024-04-06:
Hi,

So, currently, the only SoC i got using Vexii is ported to Litex.
It can run linux in single core config, i'm now working on memory coherency  / multi core, which works in sim, but i need to test on hardware + cleaning.

It is still WIP. What would you be interrested into (exactly) ?

> Is it similar to VexRiscv?

Yes, but better <3, but i'm still working on reaching feature parity.

Dolu1990, 2024-04-06:
So far, you can generate VexiiRiscv that way : 
https://spinalhdl.github.io/VexiiRiscv-RTD/master/VexiiRiscv/HowToUse/index.html#generate-verilog
There is a lot of arguments possible.
The interface generated will be the native one. Currently the only bridge implemented are toward tilelink (but not available there as an option yet, WIP)

Ultimately, bridge toward popular memory bus (wishbone axi ... ) should be added (WIP)

> VexRiscv is perfect for what I was using for, but I would like to upgrade from VexRiscv for performance

That is one of the main goal of VexiiRiscv.
I'm realy looking forward for the software / hardware D$ prefetcher, that could reaaaaly help a lot in memory intensive cases.

Dolu1990, 2024-04-06:
Note that for now, the dual issue do a bit too much in the decode / dispatch stage (all together in one cycle), there is some work there planned to split things in two stages.

closed 2025-02-12 by Dolu1990