SpinalHDL/VexiiRiscv #16: address width for xlen=64

zjin8520, 2024-06-20:
Hi,

When I use "--xlen 64 --with_mul --with_div --with_rva" to generate the 64bits VexiiRiscv core to access over 4GB memory space, I found the xxxx_logic_bus_cmd_payload_address still 32bits. Do I miss some options?  
Thank you very much!

Dolu1990, 2024-06-20:
Hi,

Current, the VexiiRiscv parameter "factory" assume the phyical address width is 32 bits,
see https://github.com/SpinalHDL/VexiiRiscv/blob/e991b315607ac8d9e2bf0af1869ef52cf0c297a4/src/main/scala/vexiiriscv/Param.scala#L337    (32)

I never tried having more than 32 bits physical address, but fondramently it should work "out of the box" if you override it to let's say 37 bits.
It may shout on your some error, in which case i need to fix things, but it should only be very minor fixes.

I'm in travel for a few more days, i can't realy look at it in details for now.
But ultimately,  an --physical-width option should be added.

zjin8520, 2024-06-21:
Hi,

I try to modify you mentioned line "case false => plugins += new memory.StaticTranslationPlugin(32)" to "case false => plugins += new memory.StaticTranslationPlugin(**xlen**) ", the verilog code is generated without any error and the payload_address changed to 64 already. but I'm not running any simulation about the generated code yet.
Thank you very much.

Dolu1990, 2024-06-21:
It should be good now, i added a --physical-width xyz arg

Having xlen driving physical width is kinda too costly as default.

closed 2024-06-21 by Dolu1990