SpinalHDL/VexiiRiscv #19: reset-vector issue

zjin8520, 2024-06-28:
Hi, 

when I use these parameters "--xlen 64 --with-mul --with-div --with-rva --with-rvZb --with-boot-mem-init --reset-vector 0x0 --physical-width 32" to generate the VexiiRisc.v,  the FetchCachelessPlugin_logic-bus_cmd_valid can't assert to fetch the code, it there something wrong? 
[image]

thanks a lot

Dolu1990, 2024-06-28:
Hi,

So far,nearly all simulation of VexiiRIscv were done using Verilator, which doesn't have x-prop simulation.
You likely have hit a xprop issue. Then if that's a bug, or a false bug i don't know.

Can you sent me your wave ? As a VCD or a FST ?

zjin8520, 2024-07-01:
Hi Sir,

I'm very sorry I can't provide the VCD format waveform immediately, but I tried to trace this problem, I found the following code, maybe it's the root cause?  Please try to analyse the code snippet, if you still need the VCD file, please let me know and I will try to generate it.  Thank you very much!
[image]
[image]

zjin8520, 2024-07-01:
Hi Sir, 

And I found another problem? When I try to use **"--reset-vector 0x8F00000000000000 --physical-width 64"** to generate RTL, I got the following error message. Actuaully, if reset-vectore's bit[63 is 1, the error will occur. Most of my job is verification, so I like to try some corner cases, sorry....

zjin8520, 2024-07-01:
Hi Sir, 

And I found another problem? When I try to use "--reset-vector 0x8F00000000000000 --physical-width 64" to generate RTL, I got the following error message. Actuaully, if reset-vectore's bit[63] is 1, the error will occur. Most of my job is verification, so I like to try some corner cases, sorry....

[image]

Jzjerry, 2024-07-01:
Hi,

> And I found another problem? When I try to use "--reset-vector 0x8F00000000000000 --physical-width 64" to generate RTL, I got the following error message. Actually, if reset-vectore's bit[63] is 1, the error will occur. Most of my job is verification, so I like to try some corner cases, sorry....

That should be because `Long` is used for `reset-vector`, so bit 63 is the sign bit of a `Long`, and the value of a `Long` should be less than +2^63.
As unsigned integers are not included in Scala, the solution might be to use `BigInt` instead, which is aligned with `PcPlugin` as well:
https://github.com/SpinalHDL/VexiiRiscv/blob/7b50ac849a11ea48083f7c2a187fd722843bccfb/src/main/scala/vexiiriscv/Param.scala#L66
to `var resetVector : BigInt = 0x80000000l`.
https://github.com/SpinalHDL/VexiiRiscv/blob/7b50ac849a11ea48083f7c2a187fd722843bccfb/src/main/scala/vexiiriscv/Param.scala#L263
to `opt[BigInt]("reset-vector")`. 
That should fix it.

Dolu1990, 2024-07-01:
Hi,

So the reason why 0x0 can't be used is that by default the PMA specification used by VexiiRiscv is : 
https://github.com/SpinalHDL/VexiiRiscv/blob/7b50ac849a11ea48083f7c2a187fd722843bccfb/src/main/scala/vexiiriscv/Param.scala#L25

Which mean, address zero is unmapped, and will always fault.

We need to add a way to parametrized the PMA specification via arguments.

> --reset-vector 0x8F00000000000000 --physical-width 64

I would say, avoid 64 bits physical memory address space, as anyway, no mmu allows such width, aswell as it create overhead in the CPU.

> opt[BigInt]("reset-vector")

Yes right :)

zjin8520, 2024-07-01:
Hi Sir,

got it, thanks for your help :)

closed 2025-02-12 by Dolu1990