SpinalHDL/VexiiRiscv #24: ti375 devkit litex

Dolu1990, 2024-09-04:
- Board : https://www.efinixinc.com/products-devkits-titaniumti375c529.html
- HDMI : https://www.digikey.com/en/products/detail/efinix-inc/EFX-HDMI/17084519?s=N4IgTCBcDaIKYDMCWA7JAPABACwCYFskQBdAXyA
- USB pmod : https://github.com/Dolu1990/pmod_usb_host_x4/blob/main/pmods.pdf

Dolu1990, 2024-11-01:
! Update !

Debian and FPGA images are uploaded here : 
https://drive.google.com/drive/folders/12WYe_igo2edwZABvF9Zta_bw4kuEbhms?usp=sharing

Here is how you can flash a sdcard using those (after you uncompressed p1 p2 : 

```
export SDCARD=/dev/???
export SDCARD_P1=${SDCARD}p1
export SDCARD_P2=${SDCARD}p2

(
echo o
echo n
echo p
echo 1
echo
echo +600M
echo y
echo n
echo p
echo 2
echo
echo +19000M
echo y
echo t
echo 1
echo b
echo p
echo w
) | sudo fdisk $SDCARD

sudo mkfs.vfat $SDCARD_P1
sudo mkfs.ext4 $SDCARD_P2
sudo e2label $SDCARD_P2 rootfs

dd if=p1.img of=$SDCARD_P1
dd if=p2.img of=$SDCARD_P2
```

silverliningeda, 2024-11-07:
excuse my ignorance as i have worked entirely on cloud FPGAs.  i am trying to collect all the hardware needed to replicate this demo.

can you add a little detail about what you mean by the USB pmod and what it is needed for ?

Dolu1990, 2024-11-07:
Hi,
The USB pmod is this extention board : 
https://github.com/Dolu1990/pmod_usb_host_x4/tree/main

It provide USB host support to the system, allowing to connect mouse, keyboard, sound card, and other USB peripherals

silverliningeda, 2024-11-07:
where can i get it? or do i need to do some soldering (which i dont know how to do) ?
thanks

Dolu1990, 2024-11-07:
To get it : 
- Either you order the PCB (i did it on pcbway), then you order the componants and mount them.
- Either you ask a compagny to do the whole process (can be done on pcbway aswell, but quite a bit more expensive)

The only thing to be careful about is "The PMOD connector is on the bottom side of the PCB in kicad, but should be solder on the top."

Dolu1990, 2024-11-07:
I would say, if you have access to a USB cable, cut the cable, solder pins on the wires and pull down resistor on it, you should be good to go. No need of a PCB (but it keeps things cleaner XD).

silverliningeda, 2024-11-07:
ok i will try to figure it out.

but even without using PMOD0 and PMOD1, it should be sufficient to use PMOD2 (Efinix HDMI extentions board) to see debian booting on a monitor right ?

Dolu1990, 2024-11-08:
The efinix HDMI extentions board doesn't use a pmod, but another connector. 

You don't event need that HDMI if you use the serial port ^^

sitestudio, 2024-12-08:
Kind of swinging around in the dark here....downloaded and extracted the img files, loaded them onto an SD card, however when it tries to boot it can't find /dev/mmcblk0p2

[    3.087246] b300        30542848 mmcblk0 ␍␊
[    3.087299]  driver: mmcblk␍␊
[    3.117643]   b301          614400 mmcblk0p1 62d912e4-01␍␊
[    3.117699] ␍␊
[    3.147040]   b302        19456000 mmcblk0p2 62d912e4-02␍␊
[    3.147095] ␍␊
[    3.176284] No filesystem could mount root, tried: ␍␊
[    3.176315]  ext3␍␊
[    3.192401]  ext2␍␊
[    3.205117]  ext4␍␊
[    3.217434]  vfat␍␊
[    3.229374]  ntfs␍␊
[    3.241068]  ntfs3␍␊
[    3.252595]  fuseblk␍␊
[    3.263961]  btrfs␍␊
[    3.275284] ␍␊
[    3.296526] Kernel panic - not syncing: VFS: Unable to mount root fs on "/dev/mmcblk0p2" or unknown-block(179,2)␍␊
[    3.326960] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 6.10.0-rc2+ #19␍␊
[    3.343336] Call Trace:␍␊
[    3.355426] [<ffffffff80004e48>] dump_backtrace+0x1c/0x24␍␊
[    3.370792] [<ffffffff8081e070>] show_stack+0x2c/0x38␍␊
[    3.385815] [<ffffffff8082dc84>] dump_stack_lvl+0x50/0x72␍␊
[    3.401103] [<ffffffff8082dcba>] dump_stack+0x14/0x1c␍␊
[    3.415908] [<ffffffff8081e24a>] panic+0xf8/0x2b0␍␊
[    3.430258] [<ffffffff80a016c2>] mount_root_generic+0x284/0x2a2␍␊
[    3.445930] [<ffffffff80a01844>] mount_root+0x164/0x180␍␊
[    3.460776] [<ffffffff80a01a5c>] prepare_namespace+0x1fc/0x258␍␊
[    3.476252] [<ffffffff80a01092>] kernel_init_freeable+0x1f8/0x214␍␊
[    3.492104] [<ffffffff8082f2ec>] kernel_init+0x1e/0x11a␍␊
[    3.506989] [<ffffffff80835fde>] ret_from_fork+0xe/0x1c␍␊
[    3.521940] SMP: stopping secondary CPUs␍␊
[    3.535473] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on "/dev/mmcblk0p2" or unknown-block(179,2) ]---␍␊

End of the SD Card imaging is:

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information: done   

1048576+0 records in
1048576+0 records out
536870912 bytes (537 MB, 512 MiB) copied, 33.9866 s, 15.8 MB/s
36132880+0 records in
36132880+0 records out
18500034560 bytes (19 GB, 17 GiB) copied, 1307.53 s, 14.1 MB/s

sitestudio, 2024-12-08:
Getting errors as it tries to boot up...downloaded the prebuilt images, and can see files in both '629MB Volume' and rootfs on the SD card (32GB):

[    5.878333] EXT4-fs (mmcblk0p2): ext4_check_descriptors: Checksum for group 123 failed (41871!=36088)␍␊
[    5.937946] EXT4-fs (mmcblk0p2): ext4_check_descriptors: Checksum for group 124 failed (25165!=19764)␍␊
[    5.997546] EXT4-fs (mmcblk0p2): ext4_check_descriptors: Checksum for group 125 failed (23813!=29303)␍␊
[    6.057181] EXT4-fs (mmcblk0p2): ext4_check_descriptors: Checksum for group 126 failed (34490!=43466)␍␊
[    6.116821] EXT4-fs (mmcblk0p2): ext4_check_descriptors: Checksum for group 127 failed (36187!=41514)␍␊
[    6.198565] EXT4-fs (mmcblk0p2): failed to initialize system zone (-117)␍␊
[    6.229913] EXT4-fs (mmcblk0p2): mount failed␍␊
[    6.259761] VFS: Cannot open root device "/dev/mmcblk0p2" or unknown-block(179,2): error -117␍␊
[    6.292696] Please append a correct "root=" boot option; here are the available partitions:␍␊
[    6.325296] b300        30542848 mmcblk0 ␍␊
[    6.325351]  driver: mmcblk␍␊
[    6.378958]   b301          614400 mmcblk0p1 39b2d565-01␍␊
[    6.379014] ␍␊
[    6.431063]   b302        19456000 mmcblk0p2 39b2d565-02␍␊
[    6.431118] ␍␊
[    6.481473] List of all bdev filesystems:␍␊
[    6.506663]  ext3␍␊
[    6.506699]  ext2␍␊
[    6.529258]  ext4␍␊
[    6.551102]  vfat␍␊
[    6.572311]  ntfs␍␊
[    6.592918]  ntfs3␍␊
[    6.612939]  fuseblk␍␊
[    6.632412]  btrfs␍␊
[    6.651515] ␍␊
[    6.686959] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,2)␍␊
[    6.711469] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 6.10.0-rc2+ #19␍␊
[    6.733724] Call Trace:␍␊
[    6.751353] [<ffffffff80004e48>] dump_backtrace+0x1c/0x24␍␊
[    6.771925] [<ffffffff8081e070>] show_stack+0x2c/0x38␍␊
[    6.791716] [<ffffffff8082dc84>] dump_stack_lvl+0x50/0x72␍␊
[    6.811392] [<ffffffff8082dcba>] dump_stack+0x14/0x1c␍␊
[    6.830170] [<ffffffff8081e24a>] panic+0xf8/0x2b0␍␊
[    6.848127] [<ffffffff80a0162a>] mount_root_generic+0x1ec/0x2a2␍␊
[    6.866987] [<ffffffff80a01844>] mount_root+0x164/0x180␍␊
[    6.884610] [<ffffffff80a01a5c>] prepare_namespace+0x1fc/0x258␍␊
[    6.902483] [<ffffffff80a01092>] kernel_init_freeable+0x1f8/0x214␍␊
[    6.920289] [<ffffffff8082f2ec>] kernel_init+0x1e/0x11a␍␊
[    6.936668] [<ffffffff80835fde>] ret_from_fork+0xe/0x1c␍␊
[    6.952642] SMP: stopping secondary CPUs␍␊
[    6.966952] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,2) ]---␍␊

Dolu1990, 2024-12-08:
Hi,

hmm i never had a error like this.
Do you have another sdcard in hand ? (another brand / generation)
To double check if the error is persisstant.

Also, can you try to check the integrity of root volume using a sdcard reader ?

Dolu1990, 2024-12-08:
Nice :D

sitestudio, 2024-12-10:
Hi,

When trying to run the abovementioned "python3 -m litex_boards.targets.efinix_ti375_c529_dev_kit ...." command, getting the following error:

/bin/sh: 1: cd: can't cd to /home/andymc/projects/TI375_Debian/build/tools/litex/pythondata-cpu-vexiiriscv/pythondata_cpu_vexiiriscv/verilog/ext/VexiiRiscv
 
The folders ext and VexiiRiscv are not there.

Is there some installation procedure missing to get them to appear?

Dolu1990, 2024-12-10:
Hi @sitestudio 
My bad, i forgot to remove the --update-repo=no from the argument list ^^

I updated the command above. (just removed --update-repo=no)
Should be all good then.

sitestudio, 2024-12-10:
Added:

default_clk_freq   = 100e6

to efinix_ti375_c529_dev_kit.py, getting further now.

sitestudio, 2024-12-10:
Hi @Dolu1990 

Now getting an error regarding USB:

USB OHCI netlist : UsbOhciWishbone_Dw32_Pc4_Pf60000000
Generating USB OHCI netlist
!!! cd /home/andymc/projects/TI375_Debian/build/tools/litex/pythondata-misc-usb_ohci/pythondata_misc_usb_ohci/verilog/ext/SpinalHDL && sbt "lib/runMain spinal.lib.com.usb.ohci.UsbOhciWishbone --port-count=4 --phy-frequency=60000000 --dma-width=32 --netlist-name=UsbOhciWishbone_Dw32_Pc4_Pf60000000 --netlist-directory=/home/andymc/projects/TI375_Debian/build/tools/litex/pythondata-misc-usb_ohci/pythondata_misc_usb_ohci/verilog"
[warn] No sbt.version set in project/build.properties, base directory: /home/andymc/projects/TI375_Debian/build/tools/litex/pythondata-misc-usb_ohci/pythondata_misc_usb_ohci/verilog/ext/SpinalHDL
[info] welcome to sbt 1.10.6 (Private Build Java 19.0.2)
[info] set current project to spinalhdl (in build file:/home/andymc/projects/TI375_Debian/build/tools/litex/pythondata-misc-usb_ohci/pythondata_misc_usb_ohci/verilog/ext/SpinalHDL/)
[error] Expected ID character
[error] Not a valid command: lib (similar: alias, plugin, client)
[error] Expected project ID
[error] Expected configuration
[error] Expected ':'
[error] Expected key
[error] Not a valid key: lib (similar: ivySbt)
[error] lib/runMain spinal.lib.com.usb.ohci.UsbOhciWishbone --port-count=4 --phy-frequency=60000000 --dma-width=32 --netlist-name=UsbOhciWishbone_Dw32_Pc4_Pf60000000 --netlist-directory=/home/andymc/projects/TI375_Debian/build/tools/litex/pythondata-misc-usb_ohci/pythondata_misc_usb_ohci/verilog
[error]    ^

Dolu1990, 2024-12-10:
Can you check in /home/andymc/projects/TI375_Debian/build/tools/litex/pythondata-misc-usb_ohci/pythondata_misc_usb_ohci/verilog/ext/SpinalHDL/project/build.properties 

?
It the file existing ?
You can update it to sbt.version=1.10.0
To see if it fixes it ?

sitestudio, 2024-12-11:
No verilog/ext/SpinalHDL/ folder tree below the pythondata_misc_usb_ohci folder so it looks like it is not setting up.

Will investigate, I should learn the Litex ecosystem :)

sitestudio, 2024-12-11:
Took out the --with-ohci and gets a bit futher to here.

CC       bios.elf
/usr/lib/gcc-cross/riscv64-linux-gnu/11/../../../../riscv64-linux-gnu/bin/ld: ../libc/libc.a(libc_ssp_chk_fail.c.o): in function `__chk_fail':
/home/andymc/projects/TI375_Debian/build/work/build/efinix_ti375_c529_dev_kit/software/libc/../../../../../tools/litex/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/ssp/chk_fail.c:13: undefined reference to `write'
collect2: error: ld returned 1 exit status
make: *** [/home/andymc/projects/TI375_Debian/build/tools/litex/litex/litex/soc/software/bios/Makefile:72: bios.elf] Error 1
rm crt0.o
make: Leaving directory '/home/andymc/projects/TI375_Debian/build/work/build/efinix_ti375_c529_dev_kit/software/bios'
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/andymc/projects/TI375_Debian/build/tools/litex/litex-boards/litex_boards/targets/efinix_ti375_c529_dev_kit.py", line 735, in <module>

Will keep playing around with it.

Dolu1990, 2024-12-11:
The OHCI thing should be fixed via this PR : 
https://github.com/enjoy-digital/litex/pull/2135

Basicaly, the repo wasn't cloned recursively XD

Dolu1990, 2024-12-11:
> /usr/lib/gcc-cross/riscv64-linux-gnu/11/../../../../riscv64-linux-gnu/bin/ld: ../libc/libc.a(libc_ssp_chk_fail.c.o): in function __chk_fail': 

Ahhh i would say this is related to the RISC-V GCC you have installed.

Looking at litex, i can see a few ways to get the RISC-V gcc : 
https://github.com/enjoy-digital/litex/commit/e148650279c18e29787e863102b1de8f0674e4e7#diff-0b495d6b6e69eaf0a0657f8daa6a92c8f838a6329ad7d378db7183d96e0f759dR357

So for instance on debian : apt install gcc-riscv64-unknown-elf

sitestudio, 2024-12-11:
All building now, a bit of a tricky path to get it working. Will write up the steps needed in my case and post here.

Thanks for your help @Dolu1990

Dolu1990, 2024-12-11:
Nice ^^
Yes please, let me know anything which isn't fixed that you had to workaround 

Cheers

esyvjlomh, 2025-01-03:
Hello @Dolu1990 
I am just try to run the commands which could generate an SoC (both in this issue and in the online documentation -- SoC -- LiteX), but seems like the commit ("ca10ab58") that the python commands use has some problems like this screenshot below:
[image]

Then I try to change the commit to the newest ("7f7c754") in core.py, and the response is like:
[image]

Maybe something in the python script should be changed?

Appreciate it if some guidance can be given!

Dolu1990, 2025-01-03:
> ca10ab58

It seems like you have a old version of litex. It should be  b4269ddc :
https://github.com/enjoy-digital/litex/blob/e3dcfbde60ee5965f4365cfdfe22cf9c7552a53a/litex/soc/cores/cpu/vexiiriscv/core.py#L159

Note there is this vexii litex argument you can use to force the repo : 
  --update-repo {latest,wipe+latest,recommended,wipe+recommended,no}
          Specify how the VexiiRiscv & SpinalHDL repo should be updated (latest: update to HEAD, recommended: Update to
          known compatible version, no: Don't update, wipe+*: Do clean&reset before checkout) (default: recommended)

esyvjlomh, 2025-01-06:
> > [ca10ab5](https://github.com/SpinalHDL/VexiiRiscv/commit/ca10ab58b4dc5b410826099bcc55c0bc30973c36)
> 
> It seems like you have a old version of litex. It should be [b4269dd](https://github.com/SpinalHDL/VexiiRiscv/commit/b4269ddcb77437143f459456eb1d1abecb0278a3) : https://github.com/enjoy-digital/litex/blob/e3dcfbde60ee5965f4365cfdfe22cf9c7552a53a/litex/soc/cores/cpu/vexiiriscv/core.py#L159
> 
> Note there is this vexii litex argument you can use to force the repo : --update-repo {latest,wipe+latest,recommended,wipe+recommended,no} Specify how the VexiiRiscv & SpinalHDL repo should be updated (latest: update to HEAD, recommended: Update to known compatible version, no: Don't update, wipe+*: Do clean&reset before checkout) (default: recommended)

That works well. Thank you so much!

juanschroeder, 2025-02-08:
Great demo, thanks.

Can you confirm I would I be able to run this in a board with 256MB of DDR3 instead of 512MB?. E.g. the cheap Kintex-7 board: https://www.aliexpress.com/item/1005006765717166.html
In the demo it looks like you have plenty of RAM available. 

Otherwise, if I prefer a Xilinx FPGA, would I be able to run it easily in a Nexys Video? This Efinix ti375 works much better than that Artix 7 one, right?

Thanks!

Dolu1990, 2025-02-08:
Hi

> would I be able to run this in a board with 256MB of DDR3?

Sure, 256 MB should be enough.

> would I be able to run it easily in a Nexys Video

Yes, it should work out of the box (via litex)

> This Efinix ti375 works much better than that Artix 7 one, right?

Right, much faster. The only specific thing is that they do not implement distributed ram. But that is fine with vexii.