SpinalHDL/VexiiRiscv #25: Add memory access statistics to probe?

Jzjerry, 2024-09-04:
Hi,

I wonder if adding memory access-related statistics to probe for better performance analysis is possible.
Maybe by counting committed load/store instructions?

Dolu1990, 2024-09-04:
Hi,

There is some infrastructure in place through the PerformanceCounterPlugin.
Here are some of the event id already implemented : 
https://github.com/SpinalHDL/VexiiRiscv/blob/dev/src/main/scala/vexiiriscv/misc/Service.scala#L31

So, additional eventId could be added, then would need to plug them into the design :)

Do you know how the RISC-V privileged performance counters works (from a privileged spec perspective) ?

Jzjerry, 2024-09-05:
> Do you know how the RISC-V privileged performance counters works (from a privileged spec perspective) ?

I haven't used them before, but I find some related code in `baremetal/hpm`. So basically it's to add performance counters and register event IDs to them, if I understand correctly.

Anyway, I will try to look into it and see if I can add an event that counts loads and stores :)

Dolu1990, 2024-09-05:
> I haven't used them before, but I find some related code in baremetal/hpm. So basically it's to add performance counters and register event IDs to them, if I understand correctly.

Yes right ^^

> Anyway, I will try to look into it and see if I can add an event that counts loads and stores :)

Let's me know how it goes, also, a few are already in place

Another thing is that if you run linux on vexii, you can also use the perf command to access them :D

Jzjerry, 2024-09-05:
It's easier than I thought haha
So I just added two more event IDs into `misc/Service.scala` and added some logic into `LsuCachelessPlugin` and `LsuL1Plugin`. 
For LsuL1 I just reused `DCACHE_ACCESS` for the load counter, and added a similar store counter:
```Scala
e.loadAccess := up.isFiring && SEL && LOAD
e.loadCount  := e.loadAccess
e.storeCount := up.isFiring && SEL && STORE
```
For LsuCacheless that didn't contain counters before, I added the above logic to the `onJoin` area.

Tested with the following code:
```C
#define LOAD_COUNT     0x1C
#define STORE_COUNT    0x1D

int data[1000];

csr_write(mhpmevent3, STORE_COUNT  );
csr_write(mhpmevent4, LOAD_COUNT   );

csr_write(mhpmcounter3, 0);
for(int i = 0;i < 1000;i++){
    data[i] = i;
}
sim_puthex(csr_read(mhpmcounter3));sim_puts("\n");
csr_write(mhpmcounter4, 0);
for(int i = 0;i < 1000;i++){
    dummy = data[i];
}
sim_puthex(csr_read(hpmcounter4));sim_puts("\n");
```
Got 1000 for both store counts and load counts with `LsuCacheless`, but 1001 load counts for `LsuL1`, I guess there is a cache miss.

Dolu1990, 2024-09-06:
Nice ^^

Yes likely a cache miss. Not that the boot assembly is likely preloading most of the data array to initialize it with zeroes

closed 2024-09-06 by Jzjerry