SpinalHDL/VexiiRiscv #27: `tselect` CSR should be always present, and hardwired if no triggers

jerryz123, 2024-09-20:
From debug spec:

> This register determines which trigger is accessible through the other trigger registers. The set of
accessible triggers must start at 0, and be contiguous.
Writes of values greater than or equal to the number of supported triggers may result in a different
value in this register than what was written. To verify that what they wrote is a valid index,
debuggers can read back the value and check that tselect holds what they wrote.

closed 2024-09-23 by Dolu1990

jerryz123, 2024-09-23:
@Dolu1990 does your change actually add the new CSRs? If (debugTriggers == 0), does this add a CSR init'd to 0? I'm still seeing the test failure.

Dolu1990, 2024-09-24:
Hi,

i just pushed some change.
- Before those change the debug triggers were only allowed for JTAG debug
- Now they can be used to generate breakpoint aswell

To enable them, you now only need 
--debug-triggers=howMany  --debug-triggers-lsu

But keep in mind, VexiiRiscv only support a subset of the triggers, mostly PC / load / store address triggers + chaining
What test are you running (which need test/need them) ?

jerryz123, 2024-09-24:
The rv64mi-p-breakpoint test.

To be clear, I don't need the rest of the trigger functionality or breakpoint functionality. The test expects that a hart which does not support those features will have tselect readable and hardwired to zero, such that sw can discover that no trigger support is available.

Dolu1990, 2024-09-24:
ok ^^ 
So,  now if you don't put --debug-triggers=howMany --debug-triggers-lsu, tselect should be hardwired to 0

commit e2a1b1ff (2024-09-23): fix #27 tselect is now always implemented