SpinalHDL/VexiiRiscv #29: How to get the area breakdown of plugins in Vexii?

Jzjerry, 2024-10-11:
Hi,

I'm trying to check the area breakdown of the plugins in Vexii (like the area of IntAlu, Lsu or Branch). 
Is there a way to generate plugins as separate verilog modules? Or are there any other ways to analyze their area?

Dolu1990, 2024-10-11:
Hi,

I would say, the only way is to run the synthesis flow between a configuration with and without a feature to get its "real" ressource usage. It may not be faisable for everything thing as there may be some coupling between a few plugins.
Using verilog module to contains logic feature get messed up a lot with hierarchy flattening.

Jzjerry, 2024-12-26:
Just made some data analysis on the impact of generation options on the area (LUTs+FFs):

[image]

Basically, I generated 1024 Vexii designs with random configurations and ran a Vivado script to synthesize them, then got the area report. For boolean options (like with-late-alu), the blue points are those that set the option to `true`, and the red ones are `false`. For integer options (like xlen and lanes), the red points are with higher values.

Could be a reference for those who want to resize their Vexii I hope 🤔.

Dolu1990, 2024-12-26:
Nice :D
Thanks ^^
1024 designs, how long did that toke XD ?

Jzjerry, 2024-12-26:
5 minutes per design, so around 4 days in total XD.
Can be faster if do it in parallel.

closed 2025-02-12 by Dolu1990