SpinalHDL/VexiiRiscv #31: FetchL1Plugin relies that cache memory is initialized

petrmikheev, 2024-10-24:
Trying to migrate my hobby project SoC from VexRiscv to VexiiRiscv. I suspect there is a bug in FetchL1Plugic:

After fetching first few commands FetchL1Plugin produces unknown (`'xxxx`) output if cache memory was not zero-initialized.

Simulation in iverilog:
[image]

After a few clock cycles `'xxxx` propagates to registers in PcPlugin and simulation gets stuck.

But if I either set `bootMemClear = true` or disable cache (`fetchL1Enable = false`) in ParamSimple, it works fine.

Dolu1990, 2024-10-24:
Hi,

bootMemClear is the intended way to avoid those x-prop.
Overall, as far i looked into the x-prop happening in Vexiiriscv (with bootMemClear = false), they all look like false bug only induced by x-prop creating impossible cases.
In other words, x-prop creating simulation only bugs.

All the test i run VexiiRiscv on, are run by Verilator, which instead of x-prop, use 1 and 0 and randomize all the values at init.

I kinda don't feel great about putting bootMemClear by default, because it consume hardware ressources and timings. Maybe instead i should put a big warning message at the very end of the VexiiRiscv hardware generation to warn about it ?

> Trying to migrate my hobby project SoC from VexRiscv to VexiiRiscv

Nice ^^ 
What SoC / memory bus are you working with ?

petrmikheev, 2024-10-24:
> In other words, x-prop creating simulation only bugs.
> All the test i run VexiiRiscv on, are run by Verilator, which instead of x-prop, use 1 and 0 and randomize all the values at init.

Thanks! Makes perfect sense then.

> Maybe instead i should put a big warning message at the very end of the VexiiRiscv hardware generation to warn about it?

Can be enough to mention it in documentation, somewhere near "Run a simulation" section. It would save me some time.

> What SoC / memory bus are you working with?

Previously used Axi4, now trying Tilelink.
SpinalHDL components for Tilelink look very nice, but it is not easy to follow all the magic that happens underneath. For example struggling to get the meaning of
```scala
// from vexiiriscv/soc/litex/Soc.scala
vexii.lsuL1Bus.setDownConnection(a = withCoherency.mux(StreamPipe.HALF, StreamPipe.FULL), b = StreamPipe.HALF_KEEP, c = StreamPipe.FULL, d = StreamPipe.M2S_KEEP, e = StreamPipe.HALF)
vexii.dBus.setDownConnection(a = StreamPipe.HALF, d = StreamPipe.M2S_KEEP)
```
and understand why it should be written exactly this way.

By the way, what is the best place to ask questions? Here? Or maybe in SpinalHDL google group?

petrmikheev, 2024-10-28:
I have a few probably stupid questions.

> setDownConnection is a way to ask the interconnect to add some pipelining stages between lsuL1Bus and the address decoder which will serve the down connections.

Do I understand right that `lsuL1Bus` is the bus from cache to memory, and when cache is enabled `dBus` is used only for non-cachable regions?
I.e.
```
     ->  fetch L1 cache -> iBus -> memory
cpu  ->  LSU L1 cache -> lsuL1Bus -> memory
     ->              dBus       -> IO regions
```

> StreamPipe.HALF is a lightweight pipeline stage which fully cut all combinatorial paths.
> StreamPipe.FULL is a "heavy weight" one which also cut all combinatorial path
> StreamPipe.M2S_KEEP is a ligtweight one which cut the valid + data path, but not the ready path

In https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Libraries/stream.html it is written that after `halfPipe()` the bandwidth divided by two. Half pipe on channel A then should mean that during all write operations it can send each beat of data only on every second clock cycle, and the bus works twice slower than it could. Seems I am missing something... Do maybe all writes go through channel C rather than A?

Also I can not figure out what `param.lsuL1Coherency` actually change. Let's take several use cases:
1) Tilelink bus with 2 masters (a. VexiiRiscv core with cache; b. video controller) and one slave (memory). Video controller makes only READ requests, but should trigger writeback if the requested data is cached and modified by the core.
2) Same as before, but add DMA controller to the same bus. It does both READ and WRITE operations (should trigger writeback & invalidate in VexiiRiscv cache), but DMA controller doesn't have its own cache.
3) Add to the same bus a second VexiiRiscv core that also has a cache.

Do all the use cases require `lsuL1Coherency` to be enabled? Or only the last one?

My current use case is the first one. I have video controller with Axi4ReadOnly interface and I am trying to figure out how to connect it to tilelink and make sure that cpu will writeback cached data when needed.

petrmikheev, 2024-10-28:
>> By the way, what is the best place to ask questions? Here? Or maybe in SpinalHDL google group?
>
> Github issue are good, this repo is good aswell ^^

I renamed the issue to better match the content :-)

Dolu1990, 2024-10-29:
Hi,

> Do I understand right that lsuL1Bus is the bus from cache to memory, and when cache is enabled dBus is used only for non-cachable regions?

Yes

> Do maybe all writes go through channel C rather than A?

Yes right for coherent caches. They send "aquire" requests on channel A (single beat), receive "aquire" responses on channel D (multi beat), and "release" dirty/clean cache lines on channel C.

> Tilelink bus with 2 masters (a. VexiiRiscv core with cache; b. video controller) and one slave (memory). Video controller makes only READ requests, but should trigger writeback if the requested data is cached and modified by the core.

Right

> Same as before, but add DMA controller to the same bus. It does both READ and WRITE operations (should trigger writeback & invalidate in VexiiRiscv cache), but DMA controller doesn't have its own cache.

Right

> Do all the use cases require lsuL1Coherency to be enabled? Or only the last one?

All cases. As soon as you have multiple masters (cpu, dma), and a given CPU has a data cache, then that CPU need lsuL1Coherency. (or it will need to do cache flush/invalidate in software)

> My current use case is the first one. I have video controller with Axi4ReadOnly interface and I am trying to figure out how to connect it to tilelink and make sure that cpu will writeback cached data when needed.

One reference for that is the litex SoC.scala.

What matter is that the CPU / DMA busses are merged together and then go into either a CacheFiber (l2), either into the HubFiber (no l2).
Those CacheFiber / HubFiber will handle all the coherency channel (a,b,c,d,e) and provide to the memory system (DDR) only the regular channel to do get/put (a,d).

petrmikheev, 2024-10-29:
Thanks for the answers!

> (or it will need to do cache flush/invalidate in software)

By the way, does VexiiRiscv support any cache flush instructions? I've seen only `fence.i` and `sfence.vma`, but they are not about flushing particular cache lines.

And how `fetch L1` interacts with `lsu L1` if lsuL1Coherency is disabled? Does `fence.i` flushes the whole `lsu L1` to ensure that `iBus` will see correct data?

> What matter is that the CPU / DMA busses are merged together and then go into either a CacheFiber (l2), either into the HubFiber (no l2).

Ah! That's what I missed!
If DMA uses only A,D channels and only simple get/put requests, will CacheFiber/HubFiber automatically generate flush/invalidate requests to CPU?

Dolu1990, 2024-10-29:
> By the way, does VexiiRiscv support any cache flush instructions? I've seen only fence.i and sfence.vma, but they are not about flushing particular cache lines.

Currently, there is nothing to flush particular cache lines. Idealy, CMO would need to be implemented to allow VexiiRiscv usages without memory coherency.
Todo ^^

> Does fence.i flushes the whole lsu L1 to ensure that iBus will see correct data?

Yes

> If DMA uses only A,D channels and only simple get/put requests, will CacheFiber/HubFiber automatically generate flush/invalidate requests to CPU?

Yes

Dolu1990, 2024-10-29:
The HubFiber will generate flush / invalidate for every request, while the CacheFiber has a dictionnary which track which cache has what, and so, will only generate usefull flush / invalidate requests

petrmikheev, 2024-10-29:
> Idealy, CMO would need to be implemented to allow VexiiRiscv usages without memory coherency.
> Todo ^^

CMO would be great! I hope it will also allow to avoid useless read-before-write requests to memory - for example when `memcpy` is going to fill the whole cache line with new data, but cpu decides to read this cache line from memory first.

> The HubFiber will generate flush / invalidate for every request, while the CacheFiber has a dictionnary which track which cache has what, and so, will only generate usefull flush / invalidate requests

Interesting. Does it mean that if I use HubFiber it can make sense to connect `iBus` around it to reduce the number of invalidate requests?

```
                DMA  ------------------> TilelinkHub -\
    / ->  lsu L1 cache   -> lsuL1Bus -/                >--- memory
CPU | ->  fetch L1 cache -> iBus ---------------------/
    \ --------------------> dBus ----> IO
```

Or `fence.i` flushes the whole `lsu L1` only if coherency is off?

Dolu1990, 2024-10-29:
>  I hope it will also allow to avoid useless read-before-write requests to memory - for example when memcpy is going to fill the whole cache line with new data, but cpu decides to read this cache line from memory first.

Ahhh that is because now, the cache is implemented as "write-back"
So, cache line will always be readed, even if the CPU only write in it. It won't go away

Write-through cache don't have that issue, but instead, they spam the memory system with little write requests.

> Interesting. Does it mean that if I use HubFiber it can make sense to connect iBus around it to reduce the number of invalidate requests?

Hmmm, if you have multiple CPU, you would then need to other CPU to manuly flush themself if they modify code sections.
But else, yes.

fence.i always flush the whole D$ + I$ for both cases.
Fondamentaly, when coherency is on, it wouldn't be necessary to flush D$

petrmikheev, 2024-10-29:
> So, cache line will always be readed, even if the CPU only write in it. It won't go away

I mean that there can be a special instruction that makes L1 cache think that the cache line is already loaded without an a actual read. I've looked through CMO spec and see that it doesn't have it. But can be a custom instruction.

```c
char *src = ..., *dst = ...;
for (int i = 0; i < size / 64; ++i) {
  fakeReadCacheLine(dst);  // custom instruction
  for (int j = 0; j < 64; j += 4) *(int*)(dst + j) = *(int*)(src + j);
  src += 64, dst += 64;
}
```

Dolu1990, 2024-10-29:
Ahhhhh lol, i didn't know about it.
Yes there could be a custom instruction for that.

petrmikheev, 2024-10-31:
I am stuck with it. Could you please help to figure out why it doesn't work?

I want to connect a blackbox to a tilelink bus. It will use only 256-byte read requests.

```scala
class VideoController extends BlackBox {
  val io = new Bundle {
    ...
    val tl_bus = master(tilelink.Bus(tilelink.BusParameter(
      addressWidth = 32,
      dataWidth    = 64,
      sizeBytes    = 256,
      sourceWidth  = 0,
      sinkWidth    = 0,
      withBCE      = false,
      withDataA    = false,
      withDataB    = false,
      withDataC    = false,
      withDataD    = true,
      node         = null
    )))
  }
  noIoPrefix()
  mapClockDomain(clock=io.clk, reset=io.reset)
}
```

```scala
  val coherent_bus = tilelink.fabric.Node().forceDataWidth(64)
  val mem_bus = tilelink.fabric.Node().forceDataWidth(64)

  val tilelink_hub = new tilelink.coherent.HubFiber()
  tilelink_hub.up << coherent_bus
  mem_bus << tilelink_hub.down

  val video_ctrl = new VideoController()
  val video_bus = tilelink.fabric.Node.down()
  val video_bus_fiber = fiber.Fiber build new Area {
    video_bus.m2s forceParameters tilelink.M2sParameters(
      addressWidth = 32,
      dataWidth = 64,
      masters = List(
        tilelink.M2sAgent(
          name = video_ctrl,
          mapping = List(
            tilelink.M2sSource(
              id = SizeMapping(0, 1),
              emits = tilelink.M2sTransfers(get = tilelink.SizeRange(256, 256))
            )
          )
        )
      )
    )
    video_bus.s2m.supported load tilelink.S2mSupport.none()
    video_bus.bus << video_ctrl.io.tl_bus
  }
  coherent_bus << video_bus  // this line causes error
```

The error (the first one, there are many of them) is
```
[error]  Error detected in phase PhaseCreateComponent
[error] ********************************************************************************
[error] ********************************************************************************
[error] 
[error] Bundle assignment is not complete. toplevel/video_bus_noDecoder_toDown_d_payload : ChannelD need 'data' but toplevel/video_bus_to_coherent_bus_up_bus_d_payload : ChannelD doesn't provide it.
[error]     spinal.lib.Stream.connectFrom(Stream.scala:317)
[error]     spinal.lib.Stream.$less$less(Stream.scala:117)
[error]     spinal.lib.Stream.$greater$greater(Stream.scala:122)
[error]     spinal.lib.bus.tilelink.Bus.$less$less(Bus.scala:418)
[error]     spinal.lib.bus.tilelink.fabric.Node$$anon$2$$anon$6.<init>(Node.scala:213)
[error]     spinal.lib.bus.tilelink.fabric.Node$$anon$2.$anonfun$noDecoder$1(Node.scala:207)
[error]     spinal.lib.bus.tilelink.fabric.Node$$anon$2.<init>(Node.scala:207)
[error]     spinal.lib.bus.tilelink.fabric.Node.$anonfun$thread$1(Node.scala:77)
```

Dolu1990, 2024-10-31:
Ahhhh i think the reason is that the memory coherency hub / l2  can't handle memory requests bigger than one cache line. So no memory request more than 64 bytes is allowed.

So either : 
- you need to tilelink.SizeRange(64)
- Would need to design a bridge to split memory 256 bytes memory accesses into 64 bytes ones, but overall, i think the best is realy to stick to "no more than 64 bytes". That allows to stream line a lot of things

petrmikheev, 2024-10-31:
Yes, changed to 64 and now it successfully generates verilog! Thanks!

By the way, is it possible to generate a more explanatory error message in such situation? No idea how would I debug it without your help.

Dolu1990, 2024-10-31:
I will take a look tomorrow ^^

The situation right now is that  Hub filter memory accesses to only allow the compatible one, which create a video DMA with no memory access at all, => no read / no write => withDataD = false

petrmikheev, 2024-10-31:
> Would need to design a bridge to split memory 256 bytes memory accesses into 64 bytes ones, but overall, i think the best is realy to stick to "no more than 64 bytes". That allows to stream line a lot of things

The only reason to use big requests was that in this case memory throughput is a bit higher. But it is not that significant.

Dolu1990, 2024-10-31:
One thing, i recently updated the documentation, in particular around the MicroSoc, you may find interresting things there, in particular : 
https://spinalhdl.github.io/VexiiRiscv-RTD/master/VexiiRiscv/Soc/microsoc.html#adding-a-custom-peripheral

I recently added a tilelink video DMA, see : 
https://github.com/SpinalHDL/VexiiRiscv/blob/dev/src/main/scala/vexiiriscv/soc/litex/Soc.scala#L152

Good luck :D

petrmikheev, 2024-10-31:
Thanks!

I used both MicroSoc and litex SoC as an examples.

Things like `demo.node at 0x10003000 of bus32` look as magic. It is really nice and convenient. But when I do something wrong and the magic breaks, it is sometimes hard to dig into details. So for now my preferred approach is to write custom modules in verilog and connect them together using SpinalHDL.

Dolu1990, 2024-10-31:
> But when I do something wrong and the magic breaks, it is sometimes hard to dig into details

Yes i agree. Error reporting isn't great when it goes into Fiber / Tilelink stuff.

For the Raw SpinalHDL things using component and so on, it should be good :D

petrmikheev, 2024-11-01:
> fence.i always flush the whole D$ + I$ for both cases.

Actually `fence.i` doesn't flush D$. At least if coherency is enabled. Checked experimentally. So I have to connect iBus to the tilelink hub rather than to memory directly.

Dolu1990, 2024-11-01:
Ahhh right, i just checked the code again, when coherency is enabled, the LsuPlugin will instead just wait that the storebuffer are drained out.

```scala
    val coherentFlusher = l1.coherency.get generate new Area{
      invalidationPorts.foreach(_.cmd.ready := withStoreBuffer.mux(storeBuffer.empty, True))
    }
```

petrmikheev, 2024-11-18:
Hi! Sometimes a have deadlocks somewhere in my SoC. I suppose they are caused by attempts to read or write invalid memory regions that are not connected to any slave interfaces. The problem is that I don't have a way to debug it: in my setup there is no integration with standard tools and no easy way for things like logic analyzer over jtag. And scenarios are too complicated for simulation (e.g. boot linux and run gcc on device).

Is there a way to add to tilelink hub a default slave interface that will be used if the address is not mapped to anything else? I want to add a module that catches all invalid requests and sends debug information via UART.

petrmikheev, 2024-11-23:
It seems that this time I faced an actual bug in VexiiRiscv.

`param.lanes = 2` causes deadlock in some cases.

Unfortunately I can't provide a simple reproducing example. The deadlock happens when I run `gcc` on device (compiling a trivial hello world). Can't simulate in iverilog - it would take days of single-thread simulation. However with `param.lanes = 1` everything works fine, and with `param.lanes = 2` when I run `gcc` there is always a deadlock (though linux boots and some other programs I tried work fine).

VexiiParams: https://github.com/petrmikheev/endeavour/blob/master/rtl/spinal/src/main/scala/endeavour/VexiiCore.scala#L40
The script I use to compile rv32 gcc: https://github.com/petrmikheev/endeavour/blob/master/scripts/build_toolchain.sh

Dolu1990, 2024-11-23:
Ahhh,
I will try dual issue on hardware, running debian to see how well it goes, and if it crash. That should be a good test case :)
Hopefully i can reproduce.

How does it crash freeze, i mean, can you reproduce easily ? is it kinda always dead lock on the same app ?

Dolu1990, 2024-11-23:
Another question is : Are all the timings passing place / route with positive slack ?

petrmikheev, 2024-11-24:
> How does it crash freeze, i mean, can you reproduce easily ? is it kinda always dead lock on the same app ?

Yes, always dead lock on the same place on every attempt. And it is definitely not just an infinite loop in software - it stopped reacting to timer interrupts.

> Another question is : Are all the timings passing place / route with positive slack ?

Timing analyzer reports that it should work fine even at 85C temperature.

Dolu1990, 2024-11-24:
In particular, which app is freezing ?

Dolu1990, 2024-11-24:
Can you send me your linux / buildroot binaries ?
Sometime, bugs only appear is specific binaries versions

petrmikheev, 2024-11-24:
> In particular, which app is freezing ?

Console `gcc`!
In particular - the binary `/usr/libexec/gcc/riscv32-unknown-linux-gnu/13.2.0/cc1`.
I build it using https://github.com/riscv/riscv-gnu-toolchain ([commit 6d7b5b7](https://github.com/riscv-collab/riscv-gnu-toolchain/commit/6d7b5b720f6c6a89b4f22c62b22c1d0ebc2dc353)).

> Can you send me your linux / buildroot binaries ?

I don't use buildroot... And an attempt to reproduce my setup in all details would require soldering.

The goal of my project was to go the whole path on my own: design fpga board, solder it, design SoC capable to run linux, write linux drivers for my custom peripherals, implement sbi and bootloader, manually bring together linux kernel and all the software (to better understand how linux works), and finally to be able to write and compile code directly on the device.

And (if `param.lanes = 1`) it works!
[image]

But reproducing the bug on your side can be a big problem. My only hope is that if you use same VexiiRiscv params, run 32-bit debian, and try to compile hello world with gcc, the freeze will happen too.

Dolu1990, 2024-11-24:
I tried some configs running 64 bits debian : 
``` 
--mmu-sync-read  --with-mul --with-div --allow-bypass-from=0 --performance-counters=0 --fetch-l1 --fetch-l1-ways=2 --lsu-l1 --lsu-l1-ways=2  --with-lsu-bypass --relaxed-branch --with-rva --with-supervisor --fetch-l1-ways=4 --fetch-l1-mem-data-width-min=64 --lsu-l1-ways=4 --lsu-l1-mem-data-width-min=64 --xlen=64 --with-rvc --with-rvf --with-rvd --fma-reduced-accuracy --fpu-ignore-subnormal --with-btb --with-ras --with-gshare --fetch-l1-hardware-prefetch=nl --fetch-l1-refill-count=3 --fetch-l1-mem-data-width-min=128 --lsu-l1-mem-data-width-min=128 --lsu-software-prefetch --lsu-hardware-prefetch rpt --performance-counters 9 --lsu-l1-store-buffer-ops=32 --lsu-l1-refill-count 4 --lsu-l1-writeback-count 4 --lsu-l1-store-buffer-slots=4 --lanes=2 --decoders=2 --with-late-alu --relaxed-div 
```

So, mainly, dual issue, with late-alu, and a few other things.
So far all seems to work well. But i forgot to try with the --with-dispatcher-buffer
Synthesis in progress XD

Dolu1990, 2024-11-24:
One thing i can see on your side, is that you only have 1 way of 4Kb for each i$ d$
This may lead to some dead lock when you have the store buffer enabled (seems it isn't your case)
But still, can you try with 4 ways for both i$d$ ?

Dolu1990, 2024-11-24:
Also, which version of linux do you use ?

petrmikheev, 2024-11-25:
> I tried some configs running 64 bits debian : ...

In your options string there are some duplicates:
```
--fetch-l1-ways=2  ... --fetch-l1-ways=4
--lsu-l1-ways=2 ... --lsu-l1-ways=4
--fetch-l1-mem-data-width-min=64 ... --fetch-l1-mem-data-width-min=128
--lsu-l1-mem-data-width-min=64 ... --lsu-l1-mem-data-width-min=128
--performance-counters=0 ... --performance-counters 9
```
I guess only the last value of each setting is used?

> Also, which version of linux do you use ?

Linux kernel 6.9.2

> This may lead to some dead lock when you have the store buffer enabled (seems it isn't your case)
> But still, can you try with 4 ways for both i$d$ ?

Tried now, unfortunately freezes anyway.

By the way, what is store buffer and when it is recommended to use it? I haven't found anything related in documentation.

Also I have questions about some other settings:

```
--fetch-l1-refill-count=3
--lsu-l1-refill-count 4 --lsu-l1-writeback-count 4
```
Is it the number of refill/writeback transaction that can be in-flight simultaneously?
Why you set `fetch-l1-refill-count` to 3? Seems I$ can request maximum 2 cache lines at the same time - the current one, and the next one if prefetcher is enabled.

```
--lsu-software-prefetch --lsu-hardware-prefetch rpt 
```
Documentation explains hardware prefetch, but what is software prefetch? Can it be useful to enable both at the same time?

Dolu1990, 2024-11-26:
Hi,

So i just tested much on ti375 fpga, running debian rv64imafdc with dual issue + late-alu + linux 6.10.0-rc2+.
Didn't had any bug so far.

Maybe the bug is somewere else than in the CPU, but was not visible with the single issue config, as things where behaving with different timings ?

> I guess only the last value of each setting is used?

Yes

> By the way, what is store buffer and when it is recommended to use it? 

I just pushed some doc about it : 
https://github.com/SpinalHDL/VexiiRiscv-RTD/commit/014aa3679a4b1bdf450d060a4cabec78b8453828
https://github.com/SpinalHDL/VexiiRiscv-RTD/commit/c49000edc8e3fe50ffe11bffb81df47eef802001

> Is it the number of refill/writeback transaction that can be in-flight simultaneously?

Yes

> Why you set fetch-l1-refill-count to 3? Seems I$ can request maximum 2 cache lines at the same time - the current one, and the next one if prefetcher is enabled.

Not always enough: 
[image]

Imagine : 
- slot 0 used as the prefetch of the next line before a jump
- slot 1 used for the cache line where you jump
- slot 2 used for the next cache line after the jump

> Documentation explains hardware prefetch, but what is software prefetch? Can it be useful to enable both at the same time?

It allows the user to explicitly prefetch a cache line via a dedicated instruction (RISC-V CMO)
It is usefull for access patterns which do not follow a regular stride (not predictable via RPT)
Typicaly some software which scan a  linked list could use this to prefetch the next element of the linked list.

petrmikheev, 2024-11-26:
> Maybe the bug is somewere else than in the CPU, but was not visible with the single issue config, as things where behaving with different timings ?

Maybe... Though I have no idea where to look. Seems that outside of the core in the dual-issue case the only visible difference should a more frequent memory requests. But it doesn't look like deadlock somewhere in memory system - when CPU freezes video controller is still able to read framebuffer and I see no artifacts on the screen.

Tried to add  `fetchL1RefillCount = 3; lsuL1RefillCount = 4; lsuL1WritebackCount = 4`, no difference.

> I just pushed some doc about it :

Thanks a lot!

>   * - --lsu-l1-store-buffer-ops=X
>     - Specify how many store miss can be pushed in the store buffer (disabled/0 by default)

So `--lsu-l1-store-buffer-ops=32` means that up to 32 `sw`/`sh`/`sb` operations can be queued if the destination is not in D$? What happens in read-after-write case (e.g. `lw` right after `sw` with the same address)? Will it search the new value in the store buffer? Or just lock while store buffer is not empty? Are there cases when store buffer can lead to slow downs on read operations?

>   * - --lsu-software-prefetch
>     - Enable RISC-V CMO for software prefetching in the D$

> It allows the user to explicitly prefetch a cache line via a dedicated instruction (RISC-V CMO)

Ah! Nice! Isn't it a partial implementation (without `prefetch.i`) of [Zicbop](https://github.com/riscv/riscv-CMOs/blob/master/cmobase/Zicbop.adoc) extension? If I understand correctly it can be enabled in gcc by adding `_zic64b_zicbop` to `-march` string ([gcc options](https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Options.html)). Somehow from your previous messages I thought that no cache-related instructions are implemented yet.

Maybe mention in documentation that it is `Zicbop`? And probably implement `prefetch.i` (can be no-op) - to prevent crashing if gcc emits it when zicbop is enabled.

And a feature request from me: would be great to have [`cbo.zero`](https://github.com/riscv/riscv-CMOs/blob/master/cmobase/Zicboz.adoc). It is almost the same that I mentioned [above](https://github.com/SpinalHDL/VexiiRiscv/issues/31#issuecomment-2445149030) and it seems to be easy to add. I think (not sure if compilers can do it right now) that aside of `memset` it can speedup `memcpy` up to 30% - by avoiding reading memory at destination location.

Dolu1990, 2024-11-26:
> Maybe... Though I have no idea where to look. Seems that outside of the core in the dual-issue case the only visible difference should be a more frequent memory requests. But it doesn't look like deadlock somewhere in memory system - when CPU freezes video controller is still able to read framebuffer and I see no artifacts on the screen.

Weird.

Did you tried to inspect the CPU state via openocd + jtag ?
To see if the CPU is deadlocked in a hard way, or if there is a software lockup ?
That could already rule out a lot :)

> So --lsu-l1-store-buffer-ops=32 means that up to 32 sw/sh/sb operations can be queued if the destination is not in D$?

Yes

> What happens in read-after-write case

Then the load will be cancel, then the CPU will retry and hopefully successed. else retry again and again

> Will it search the new value in the store buffer?

No, to expensive in fpga hardware ^^
I guess on ASIC they would do that.

> Or just lock while store buffer is not empty?

it will retry again and again (lazy implementation)

>  Are there cases when store buffer can lead to slow downs on read operations?

Hmmm
Maybe a bit for miss on "store X" followed by "load X", but probably not a big deal.

> Ah! Nice! Isn't it a partial implementation (without prefetch.i) of Zicbop

right.

> And a feature request from me: would be great to have [cbo.zero](https://github.com/riscv/riscv-CMOs/blob/master/cmobase/Zicboz.adoc). It is almost the same that I mentioned https://github.com/SpinalHDL/VexiiRiscv/issues/31#issuecomment-2445149030 and it seems to be easy to add. I think (not sure if compilers can do it right now) that aside of memset it can speedup memcpy up to 30% - by avoiding reading memory at destination location.

Yes right, it would probably help a lot for those operation.
It isn't realy planned so far, have a lot of other pending things XD
But for sure it would be nice to have.

petrmikheev, 2024-12-02:
> Did you tried to inspect the CPU state via openocd + jtag ?
> To see if the CPU is deadlocked in a hard way, or if there is a software lockup ?
> That could already rule out a lot :)

I've tried it now. Yes, it is deadlocked in a hard way :(

```
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
0xc026f910 in ?? ()
(gdb) x/10i ($pc-8)                     // just to illustrate that debugger works
   0xc026f908:	fence	i,ir            // it is not the code that freezes
   0xc026f90c:	srli	a4,a5,0xb
=> 0xc026f910:	andi	a4,a4,521
   0xc026f914:	bnez	a4,0xc026f902
   0xc026f916:	addi	s5,s5,1
   0xc026f918:	xor	s3,s3,s2
   0xc026f91c:	bne	s4,s5,0xc026f8c6
   0xc026f920:	mv	s8,s3
   0xc026f922:	lw	a6,-88(s0)
   0xc026f926:	lw	a2,-92(s0)
(gdb) c
Continuing.
^Cunable to halt hart 0                 // here I tried to interrupt after the freeze
  dmcontrol=0x00000001
  dmstatus =0x00430c82
Could not read registers; remote failure reply 'E0E'
(gdb)
```

Does the EmbeddedRiscvJtag/openocd/gdb chain support anything like "run next N instructions and stop"? Then I would try to find the place it freezes with a binary search. When I try `si` it works the same as `c` - just continues evaluation until I manually interrupt it.

Dolu1990, 2024-12-02:
hmm going with GDB is a bad idea in this kind of case
Better go with a more lightweight telnet localhost 4444 directly to openocd, and then from there : 
halt
reg pc
step
reg pc

things like that. It has much less side effect than GDB

> run next N instructions and stop"

Maybe the openocd "step" instruction has a count argument ? i don't know

Could you try with openocd telnet ? To avoid the gdb noise.

In particular, if you can retreive which is faulty PC, can you send me your vmlinux  elf binary ?

Dolu1990, 2024-12-02:
Do you handle unaligned load/store emulation in your SBI ?

petrmikheev, 2024-12-02:
No, my sbi trap handler just prints error message in case of unaligned access. Here it is not the case. I suspect that there is some specific sequence of instructions that can lead to hardware deadlock in dual-issue VexiiRiscv. Need only to find the exact place it freezes to check it.

Dolu1990, 2024-12-02:
Let me know how things goes, i'm realy trying to get VexiiRiscv as bug free as possible.
Also, if we have to probe things in order to progress i may have a few signals which could help

Dolu1990, 2024-12-02:
Also, note that having a setup where you can reproduce a bug in a non-random way, is valuable, backup all the binaries / elf files :D

petrmikheev, 2024-12-04:
Let's try to reproduce it on your side.

Here is 40MB initrd image that contains ext2 filesystem with rv32 `cc1` binary and a few libraries it needs: [cc1.initrd](https://cloud.mikhe.ch/index.php/s/7BUwy02OG0n7jlx).

Linux kernel is not included because I use a custom uart controller with custom uart driver which will not work with any other hardware.

I hope this will be enough to reproduce the freeze:
- VexiiRiscv rv32imacsu with param.lanes = 2
- At least 128MB RAM
- More or less any rv32 linux kernel with ext2 support and `BLK_DEV_RAM=y` `BLK_DEV_RAM_SIZE >= 40MB`
- Copy initrd to RAM and start the kernel with "root=/dev/ram" in bootargs and proper initrd-start/initrd-end values.

<details>
  <summary>My DTS (all peripherals except UART are disabled)</summary>

  ```
  // SPDX-License-Identifier: GPL-3.0
  /dts-v1/;

  / {
    #address-cells = <1>;
    #size-cells = <1>;
    model = "Endeavour";

    chosen {
      bootargs = "console=ttyS0,sbi:115200e earlycon=endeavour_sbi debug root=/dev/ram";
      linux,initrd-start = <0x85000000>;
      linux,initrd-end   = <0x87800000>;
    };

    cpus {
      #address-cells = <1>;
      #size-cells = <0>;
      timebase-frequency = <1000000>;
      cpu0: cpu@0 {
        device_type = "cpu";
        reg = <0>;
        status = "okay";
        compatible = "riscv";
        riscv,isa-base = "rv32i";
        riscv,isa-extensions = "i", "m", "a", "c", "zicntr", "zicsr", "zifencei";
        mmu-type = "riscv,sv32";
        intc0: interrupt-controller {
          #interrupt-cells = <1>;
          interrupt-controller;
          compatible = "riscv,cpu-intc";
        };
      };
    };

    plic: interrupt-controller@4000000 {
      compatible = "riscv,plic0";
      reg = <0x4000000 0x4000000>;
      #address-cells = <0>;
      #size-cells = <0>;
      #interrupt-cells = <1>;
      interrupt-controller;
      interrupts-extended = <&intc0 9>;
      riscv,ndev = <4>;
    };

    uart: serial@1000 {
      compatible = "endeavour,uart";
      reg = <0x1000 12>;
      interrupts-extended = <&plic 1>;
    };

    memory@80000000 {
      device_type = "memory";
      reg = <0x80000000 0x08000000>;
    };

    reserved-memory {
      #address-cells = <1>;
      #size-cells = <1>;
      ranges;
      framebuffer@80000000 {
        no-map;
        reg = <0x80000000 0x400000>;
      };
    };

  };
  ```

</details>

`/sbin/init` in this image is sh script:
```sh
#!/bin/sh

mount -t sysfs none /sys
mount proc /proc -t proc
mount -o remount,noatime,rw /   # remount initrd as rw, otherwise cc1 can't create output file
/cc1 -v -o test.s test.c       # freeze happens here
echo OK
```

In my case the output is the following:
```
...
[    2.790347] RAMDISK: ext2 filesystem found at block 0
[    2.790775] RAMDISK: Loading 40960KiB [1 disk] into ram disk...  
[    4.874120]  
[    7.597555]  
[   10.310734]  
[   13.510638]  
[   16.388365] done.
[   17.285606] VFS: Mounted root (ext2 filesystem) readonly on device 1:0.
[   17.286655] devtmpfs: error mounting -2
[   17.288953] Freeing unused kernel image (initmem) memory: 132K
[   17.289409] Kernel memory protection not selected by kernel config.
[   17.289889] Run /sbin/init as init process
[   17.290206]   with arguments:
[   17.290455]     /sbin/init
[   17.290705]   with environment:
[   17.290955]     HOME=/
[   17.291184]     TERM=linux
ignoring nonexistent directory "/home/petya/rv32imac-linux-toolchain/native/lib/gcc/riscv32-unknown-linux-gnu/13.2.0/include"
ignoring nonexistent directory "/usr/local/include"
ignoring nonexistent directory "/home/petya/rv32imac-linux-toolchain/native/include"
ignoring nonexistent directory "/home/petya/rv32imac-linux-toolchain/native/lib/gcc/riscv32-unknown-linux-gnu/13.2.0/include-fixed"
ignoring nonexistent directory "/home/petya/rv32imac-linux-toolchain/native/riscv32-unknown-linux-gnu/include"
ignoring nonexistent directory "/home/petya/rv32imac-linux-toolchain/native/include"
#include "..." search starts here:
#include <...> search starts here:
End of search list.
cc1: error: no include path in which to search for stdc-predef.h
 main
Analyzing compilation unit      <= after this line it freezes and doesn't print OK
```

I hope it will not be just a waste of your time. I checked all that I could and everything looks like a deadlock in VexiiRiscv, but given the stupidity of some of my bugs I will not be very surprised if something is wrong on my side.

Dolu1990, 2024-12-05:
> Linux kernel is not included because I use a custom uart controller with custom uart driver which will not work with any other hardware.

Hmm, but i could use the SBI putc/getc instead, isn't it (hvc0)?
As long as it is enabled in the linux kernel.

petrmikheev, 2024-12-05:
> Hmm, but i could use the SBI putc/getc instead, isn't it (hvc0)?
> As long as it is enabled in the linux kernel.

I had it disabled, but now has enabled and rebuilt the kernel. [Image.gz](https://github.com/user-attachments/files/18026868/Image.gz)

petrmikheev, 2024-12-25:
```
[Progress] Verilator compilation started
[info] Found cached verilator binaries
[Progress] Verilator compilation done in 350.358 ms
[Progress] Start MinimalLinuxSim test simulation with seed 42
Loading sbi.bin DONE
Loading /home/petya/endeavour-tools/linux_kernel/arch/riscv/boot/Image DONE
Loading /home/petya/endeavour-tools/cc1.initrd DONE
Start
[SBI] Probing extension 0x735049
[SBI] Probing extension 0x52464E43
[    0.000000] Linux version 6.9.2 (petya@petya-ThinkPad-P52) (riscv32-unknown-linux-gnu-gcc (gc891d8dc23e) 13.2.0, GNU ld (GNU Binutils) 2.42) #27 Thu Dec  5 17:51:31 CET 2024
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80400000
[    0.000000] Machine model: Endeavour
[    0.000000] SBI specification v0.2 detected
[    0.000000] SBI implementation ID=0x0 Version=0x0
[    0.000000] SBI TIME extension detected

... 10 hours later ...

[   16.499706] Run /sbin/init as init process
[   16.499997]   with arguments:
[   16.500222]     /sbin/init
[   16.500449]   with environment:
[   16.500680]     HOME=/
[   16.500893]     TERM=linux
ignoring nonexistent directory "/home/petya/rv32imac-linux-toolchain/native/lib/gcc/riscv32-unknown-linux-gnu/13.2.0/include"
ignoring nonexistent directory "/usr/local/include"
ignoring nonexistent directory "/home/petya/rv32imac-linux-toolchain/native/include"
ignoring nonexistent directory "/home/petya/rv32imac-linux-toolchain/native/lib/gcc/riscv32-unknown-linux-gnu/13.2.0/include-fixed"
ignoring nonexistent directory "/home/petya/rv32imac-linux-toolchain/native/riscv32-unknown-linux-gnu/include"
ignoring nonexistent directory "/home/petya/rv32imac-linux-toolchain/native/include"
#include "..." search starts here:
#include <...> search starts here:
End of search list.
[01m[Kcc1:[m[K [01;31m[Kerror: [m[Kno include path in which to search for stdc-predef.h
 main
Analyzing compilation unit
[Error] Simulation failed at time=18854063680
[error] spinal.sim.SimFailure: Vexii hasn't commited anything for too long, waiting on uop 0xBC69
[error] 	at spinal.core.sim.package$.simFailure(package.scala:214)
[error] 	at vexiiriscv.test.VexiiRiscvProbe.$anonfun$checkCommits$2(VexiiRiscvProbe.scala:553)
[error] 	at vexiiriscv.test.VexiiRiscvProbe.$anonfun$checkCommits$2$adapted(VexiiRiscvProbe.scala:547)
[error] 	at scala.collection.TraversableLike$WithFilter.$anonfun$foreach$1(TraversableLike.scala:985)
[error] 	at scala.collection.IndexedSeqOptimized.foreach(IndexedSeqOptimized.scala:36)
[error] 	at scala.collection.IndexedSeqOptimized.foreach$(IndexedSeqOptimized.scala:33)
[error] 	at scala.collection.mutable.ArrayOps$ofRef.foreach(ArrayOps.scala:198)
[error] 	at scala.collection.TraversableLike$WithFilter.foreach(TraversableLike.scala:984)
[error] 	at vexiiriscv.test.VexiiRiscvProbe.checkCommits(VexiiRiscvProbe.scala:547)
[error] 	at vexiiriscv.test.VexiiRiscvProbe.$anonfun$new$2(VexiiRiscvProbe.scala:630)
[error] 	at spinal.core.sim.package$SimClockDomainPimper.$anonfun$onSamplings$2(package.scala:1111)
[error] 	at spinal.core.sim.package$SimClockDomainPimper.$anonfun$onSamplings$2$adapted(package.scala:1111)
[error] 	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
[error] 	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
[error] 	at spinal.core.sim.package$SimClockDomainPimper.$anonfun$onSamplings$1(package.scala:1111)
[error] 	at spinal.core.sim.package$$anon$1.update(package.scala:248)
[error] 	at spinal.sim.SimManager.runWhile(SimManager.scala:340)
[error] 	at spinal.sim.SimManager.runAll(SimManager.scala:262)
[error] 	at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:614)
[error] 	at spinal.core.sim.SimCompiled.doSimUntilVoid(SimBootstraps.scala:587)
[error] 	at b3.MinimalLinuxSim$.delayedEndpoint$b3$MinimalLinuxSim$1(MinimalLinuxSim.scala:92)
[error] 	at b3.MinimalLinuxSim$delayedInit$body.apply(MinimalLinuxSim.scala:89)
[error] 	at scala.Function0.apply$mcV$sp(Function0.scala:39)
[error] 	at scala.Function0.apply$mcV$sp$(Function0.scala:39)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
[error] 	at scala.App.$anonfun$main$1$adapted(App.scala:80)
[error] 	at scala.collection.immutable.List.foreach(List.scala:431)
[error] 	at scala.App.main(App.scala:80)
[error] 	at scala.App.main$(App.scala:78)
[... 203 lines cut ...]
    sim.compile(new MinimalLinuxSim).doSimUntilVoid("test", seed = 42){dut =>
      val probe = new VexiiRiscvProbe(cpu = dut.cpu.core.logic.core, kb = None)
      probe.autoRegions()

      dut.loadToMem(0, "sbi.bin")
      dut.loadToMem(0x400000l, "/home/petya/endeavour-tools/linux_kernel/arch/riscv/boot/Image")
      dut.loadToMem(0x5000000l, "/home/petya/endeavour-tools/cc1.initrd")

      dut.clockDomain.forkStimulus(20)

      disableSimWave()
      delayed(18853000000L)(enableSimWave())

      dut.listenUart()
    }
  }
  ```

</details>

- [sbi.bin](https://cloud.mikhe.ch/index.php/s/njCz4KvJ0Pe001y)
- [Image](https://cloud.mikhe.ch/index.php/s/9SuanIWvrYX8mpm)
- [cc1.initrd](https://cloud.mikhe.ch/index.php/s/7BUwy02OG0n7jlx)

In 9 hours I'll upload dump of last 1ms of the simulation.

petrmikheev, 2024-12-25:
[wave.vcd](https://cloud.mikhe.ch/index.php/s/ikTfYxA39sYDvIb)

[image]

closed 2024-12-26 by Dolu1990

Dolu1990, 2024-12-26:
Nice :D :D :D
GG for the sim setup and capture.

The bug was reproducable in sim via this sequance : 

```asm
  la x1, data
  nop  // Ensure the memory alignment of the following assembly, to be sure the decoder handle beq and mulh the same cycle
  lw x2, 0(x1) // Generate a late x2 value, this will ensure the beq is scheduled to a late ALU
  beq x2, x2, 1f //That late alu BEQ is taken, which will flush the mulh
  mulh x3, x0, x0 //CPU freeze as the mulh logic which halt the CPU one extra cycle (to help with timeings) was buggy when the instruction is canceled.
```

A fix is pushed in VexiiRiscv dev

:D THANKS for your perseverance :D

petrmikheev, 2024-12-26:
Thanks! Probably testing VexiiRiscv is the main practical benefit of my project :-)
I am glad that it didn't turn out to be another stupid bug on my side.

commit 823054e3 (2024-12-26): fix #31 relaxed mul freezing the CPU. GG petrmikheev <3