SpinalHDL/VexiiRiscv #32: terminal reports error when using sbt "Test/runMain vexiiriscv.Generate" generates verilog

littlezpf666, 2024-11-15:
My Linux verilog generating environment is sbt 1.10.0 (Red Hat, Inc. Java 11.0.23). when using sbt "Test/runMain vexiiriscv.Generate" to generate verilog, the terminal reports the error information as below:

[error] java.lang.RuntimeException: Nonzero exit value: 129
[error] 	at scala.sys.package$.error(package.scala:30)
[error] 	at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.slurp(ProcessBuilderImpl.scala:138)
[error] 	at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.$bang$bang(ProcessBuilderImpl.scala:108)
[error] 	at $ceb55e376969a594a027$.liftedTree1$1(build.sbt:91)
[error] 	at $ceb55e376969a594a027$.gitHash(build.sbt:90)
[error] 	at $ceb55e376969a594a027$.$anonfun$core$8(build.sbt:155)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:63)
[error] 	at sbt.std.Transform$$anon$4.work(Transform.scala:69)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:283)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:24)
[error] 	at sbt.Execute.work(Execute.scala:292)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:283)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:265)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:65)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
[error] 	at java.base/java.lang.Thread.run(Thread.java:829)
[error] (ProjectRef(uri("file:/public/Vexii/VexiiRiscv/ext/SpinalHDL/"), "core") / Compile / managedSources) Nonzero exit value: 129
[error] Total time: 19 s, completed Nov 14, 2024, 3:52:22 PM

Is this the problem of my sbt and java version or something else?

Dolu1990, 2024-11-15:
Hi, it seems to fail in   s"git -C ${dir.toString} rev-parse HEAD".!!

Do you have git installed ?

littlezpf666, 2024-11-15:
Yes, I've installed the git version 1.8.3.1

Dolu1990, 2024-11-15:
I just tried : 

```shell
git clone --recursive https://github.com/SpinalHDL/VexiiRiscv.git
cd VexiiRiscv
sbt "Test/runMain vexiiriscv.Generate"
``` 

All went smooth on my side. So probably the issue is environnement related.

The failling line of code is here : 
https://github.com/SpinalHDL/SpinalHDL/blob/f17e096a004d07e17ab1d8e9c44847df4b8a261c/build.sbt#L91

Can you try to run "(cd ext/SpinalHDL && git -C . rev-parse HEAD)" in a terminal ? to see if it is happy ?

Else, replace s"git -C ${dir.toString} rev-parse HEAD".!!  by "chocolate" ^^

littlezpf666, 2024-11-15:
when I try to run cd ext/SpinalHDL && git -C . rev-parse HEAD, terminal is unhappy. 
[pengfei@LSHEMBD18 VexiiRiscv]$ cd ext/SpinalHDL && git -C . rev-parse HEAD
Unknown option: -C
usage: git [--version] [--help] [-c name=value]
           [--exec-path[=<path>]] [--html-path] [--man-path] [--info-path]
           [-p|--paginate|--no-pager] [--no-replace-objects] [--bare]
           [--git-dir=<path>] [--work-tree=<path>] [--namespace=<name>]
           <command> [<args>]

Dolu1990, 2024-11-15:
Ahhhhh XDXDXD
So on my system, the -C option for git is upper case, while on your it is lower case XDXD

Can you try with this fix for here, and let me know if that good for you ? https://github.com/SpinalHDL/SpinalHDL/blob/f17e096a004d07e17ab1d8e9c44847df4b8a261c/build.sbt#L89  ? 

```scala
import sys.process._
def gitHash(dir: File) = (try {
  s"git -C ${dir.toString} rev-parse HEAD".!!
} catch{
  case e : Throwable => "???"
}).linesIterator.next()
```

littlezpf666, 2024-11-15:
I've tried your fix. And succeed in running sbt "Test/runMain vexiiriscv.Generate" XDXD. It's good for me.
and I also tried only to modify git -C with git -c. Verilog can also be generated successfully.
Thanks for your help

Dolu1990, 2024-11-15:
Thanks ^^
I will pushed the fix in SpinalHDL

commit 0c134417 (2024-11-15): fix #32 update SpinalHDL
closed 2024-11-15 by Dolu1990