SpinalHDL/VexiiRiscv #38: About Regression in the test folder

esyvjlomh, 2024-12-25:
Hello,

I tried to run the object "RegressionSingle" in"src/test/scala/vexiiriscv/tester/Regression.scala" which seems like the similar regression test in Vex and Nax. The flow processed just well, but when I open the related fst wave in the related folder (for example "VexiiRiscv/regression/VexiiRiscv_557671459_tests/riscv-arch-test/rv32i_m/M/divu-01/wave.fst"), it do not contain any waves.

[image]

The file is only 44KB btw, which I think is not reasonable.

I tried to add some parameters and run it in both IntelliJ IDEA and sbt in terminal, but the parameters which are used in the Self Contained Tutorial cannot be reused here, and I cannot extract the wave still.

Maybe the regression flow is still WIP? If I want to extract these waves, how should I modify the code? 

Appreciate it if some guidance can be given!

Dolu1990, 2024-12-27:
Hi,

The testbench always run verilator with the wave generation code in, but disable the capture of the wave by default. This allows to have the same behaviour between a simulation without wave and with wave (verilator run differently when the wave generation code in is disabled).

I push in dev the possibility to add --trace-all as an command line argument to ask capturing the wave and a few other logs
Should work well with it ^^

Dolu1990, 2024-12-27:
> Maybe the regression flow is still WIP? If I want to extract these waves, how should I modify the code?

So, normaly, i'm not asking the regression flow to generate waves, because it can get big realy fast.
instead, i'm using the text files "args" generated in the regression folder to collect the regular vexiiriscv.tester.TestBench commands to rerun a specific test, and there add --trace-all

commit 9ae7c9a2 (2024-12-27): fix #38 regression --trace-all supported
closed 2024-12-27 by Dolu1990