SpinalHDL/VexiiRiscv #41: What is the right way to implement bootloader ROM?

petrmikheev, 2025-03-01:
What I don't understand in MicroSoc / litex Soc examples is how the SoC gets the first command to execute.

As I see reset vector is set to 0x80000000 - the beginning of main memory. It is clear how it works in simulation - simply loading a file into underlying memory: `elf.load(dut.system.ram.thread.logic.mem, 0x80000000l, true)`. But how it is supposed to work on a real device where the main memory is empty on boot?

## Approach 1

In my project I use a very hackish implementation of embedded bootloader which can initialize sdcard and load programs from there to main memory:
```scala
  val internalRam = new tilelink.fabric.RamFiber(internalRamSize)
  fiber.Handle { internalRam.thread.logic.mem.generateAsBlackBox() }
  internalRam.up at (internalRamBaseAddr, internalRamSize) of mem_bus
  ...
  ramBridge.up at (externalRamBaseAddr, externalRamSize) of mem_bus
```
(full code [here](https://github.com/petrmikheev/endeavour/blob/master/rtl/spinal/src/main/scala/endeavour/EndeavourSoc.scala#L178))

```verilog
module Ram_1wrs(
  input clk, input en, input wr,
  input [10:0] addr,
  input [7:0] mask,
  input [63:0] wrData,
  output reg [63:0] rdData
);

OnChipRAM internal_ram( // FPGA memory block, initialized with bootloader code
        .address(addr),
        .byteena(mask),
        .clken(en),
        .clock(clk),
        .data(wrData),
        .wren(wr & addr[10]),  // first 8KB is ROM, ignore write attempts
        .q(rdData));
endmodule
```

I think that my solution with overriding `Ram_1wrs` is awful, but can't figure out what is the right way to connect a bootable ROM to VexiiRiscv.

## Approach 2

Also tried to have a very simple ROM implementation connected to APB bus:
```verilog
module MyROM (
  input reset, input clk,

  input       [9:0] apb_PADDR,
  input             apb_PSEL,
  input             apb_PENABLE,
  output            apb_PREADY,
  input             apb_PWRITE,
  input      [31:0] apb_PWDATA,
  output     [31:0] apb_PRDATA
);

  reg [31:0] mem [255:0];
  initial $readmemh("rom.mem", mem);

  assign apb_PREADY = 1'b1;
  reg [8:0] addr;
  assign apb_PRDATA = mem[addr];
  always @(posedge clk) addr <= apb_PADDR[9:2];
endmodule
```

It works well with FetchCachelessPlugin, but FetchL1Plugin can't read from there (I suppose because burst transactions are not supported in APB bridge).

## Questions

1) What is the easiest way to implement a small bootable ROM?

- I'd like to keep things simple and not to implement tilelink burst transactions manually.
- Latency of the ROM is not important - it will be read only once on startup. It is more important to minimize resource usage.
- Better if the ROM uses only a single FPGA memory block (note: 32 bit tilelink.fabric.RamFiber takes at least 4, one for each byte).

2) I am a bit surprised that I can't find anything related in documentation. Feels like I miss something obvious. How do people usually load initial bootloader code to VexRiscv/VexiiRiscv?

petrmikheev, 2025-03-01:
> That SoC is designed for FPGA, where RAM can get preloaded by the bitstream itself.

Oh, I see. It is `p.ramElf.foreach(new Elf(_, p.vexii.xlen).init(ram.thread.logic.mem, 0x80000000l))` in [MicroSoc.scala](https://github.com/SpinalHDL/VexiiRiscv/blob/dev/src/main/scala/vexiiriscv/soc/micro/MicroSoc.scala#L83), I somehow haven't noticed this line at first.

> Do you target FPGA or ASIC?

FPGA. But in my case the main memory is a separate DDR chip. So I had to add an additional internal memory just to put there the bootloader code.

It is needed only for startup and I wanted to make it read-only to ensure that soft reset (i.e. VexiiRiscv reset, not FPGA reinitialization) will return the system to the initial state even after accidental memory write to a wrong location.

My solution (see "approach 1" above) works, but there are two things I don't like about it:

1) The way I made it read-only via overriding Ram_1wrs is ... Well, doesn't look like a good code. At least because it relies on the fact that there is only one blackboxed instance of Ram_1wrs. I wonder what would be a proper SpinalHDL way to implement ROM.

2) Resource usage is suboptimal. In order to allow byte-enable (which is not needed for ROM anyway) SpinalHDL generates a separate array for each byte of memwidth. So if I have 64 bit bus, tilelink.fabric.RamFiber uses at least 8 1KB memory blocks even if only a few hundred bytes needed. The bootloader runs once, and then the memory blocks remain unused. I'd like to save them for something more useful, e.g. for bigger caches. Also I suspect that having one more cacheable memory region (RamFiber tags the memory as cacheable) and one more slave port on memory bus makes the interconnect more complicated.

What I want is to have a small 1KB on-chip bootable memory preloaded by the bitstream, using as little memory and logic resources as possible. It doesn't need to be cacheable, doesn't need to support writes, can be slow if it simplifies implementation. If you have any advice how to implement it, please tell :)

Dolu1990, 2025-03-05:
Hmm there is no ROM implemented with tilelink so far, but it is very easy to do, just copy the current tilelink.fabric.RamFiber implementation and remove the write path. Don't tweek the Ram_1wrs i would say.

petrmikheev, 2025-03-22:
OK, I implemented something. If `content` is small, it uses only one memory block in FPGA, regardless of bus width. Takes less logic cells than `RamFiber + ram.up.forceDataWidth(16)`. Posting it here, for the case if somebody will have a similar use case.

```scala
// ROM optimized for low resource usage in FPGA.
// Has memory width 16 bit and bus width >= 32 bit. Takes several cycles per beat.
class SlowRom(np : tilelink.NodeParameters, val content : Array[Byte]) extends Component {
  val io = new Bundle{
    val up = slave port tilelink.Bus(np)
  }

  val p = io.up.p
  val memoryWidthBytes = 2
  val memoryWidth = memoryWidthBytes * 8
  val memoryDepth = (1<<p.addressWidth)/memoryWidthBytes
  println(s"ROM[${memoryDepth} * ${memoryWidth} bytes] <= content ${content.length} bytes")

  val reshapedContent = new Array[Bits](memoryDepth)
  for (i <- 0 to memoryDepth - 1) {
    var v = BigInt(0)
    for (j <- (i+1)*memoryWidthBytes-1 to i*memoryWidthBytes by -1) {
      if (j < content.length) v = (v << 8) | (content(j).toInt & 0xff)
    }
    reshapedContent(i) = v
  }

  val mem = Mem.fill(memoryDepth)(Bits(memoryWidth bits)) init(reshapedContent)

  val stepCount = p.dataWidth / memoryWidth
  val data = Reg(Bits(p.dataWidth-memoryWidth bits))
  val size = Reg(UInt(p.sizeWidth bits))
  val source = Reg(UInt(p.sourceWidth bits))
  val beatCounter = Reg(UInt(math.max(p.beatWidth, 1) bits))
  val stepCounter = Reg(UInt(log2Up(stepCount + 1) bits))
  val busy = Reg(Bool()) init(False)
  val stepEnable = busy && (io.up.d.ready || (stepCounter =/= 0))
  if (p.beatWidth == 0) beatCounter := 0

  val memAddress = Reg(mem.addressType)
  val memAddressLow = memAddress((log2Up(p.sizeBytes / memoryWidthBytes) - 1) downto 0)
  val memData = mem.readSync(memAddress, stepEnable)

  io.up.a.ready := !busy
  io.up.d.valid := busy && (stepCounter === 0)
  io.up.d.opcode := tilelink.Opcode.D.ACCESS_ACK_DATA
  io.up.d.param := 0
  io.up.d.source := source
  io.up.d.size := size
  io.up.d.denied := False
  io.up.d.corrupt := False
  io.up.d.data := memData ## data

  when (io.up.a.fire) {
    source := io.up.a.source
    size := io.up.a.size
    memAddress := io.up.a.address >> log2Up(memoryWidthBytes)
    if (p.beatWidth > 0) beatCounter := io.up.a.sizeToBeatMinusOne
    stepCounter := stepCount
    busy := True
  }
  when (stepEnable) {
    data := memData ## (data >> memoryWidth)
    memAddressLow := memAddressLow + 1
    stepCounter := stepCounter - 1
    when (stepCounter === 0) {
      stepCounter := stepCount - 1
      if (p.beatWidth > 0) beatCounter := beatCounter - 1
      when (beatCounter === 0) { busy := False }
    }
  }
}

class SlowRomFiber(val content : Array[Byte]) extends Area {
  val up = tilelink.fabric.Node.up()
  up.addTag(PMA.EXECUTABLE)
  val addressWidth = log2Up(content.length)
  val thread = fiber.Fiber build new Area{
    up.m2s.supported load up.m2s.proposed.intersect(tilelink.M2sTransfers(get=tilelink.SizeRange(4, 64))).copy(addressWidth = addressWidth)
    up.s2m.none()
    val logic = new SlowRom(up.bus.p.node, content)
    logic.io.up << up.bus
  }
}
```

closed 2025-03-22 by petrmikheev