SpinalHDL/VexiiRiscv #52: Running Executable that takes data from a file (CMD line args)

nishanthsmurthy24, 2025-04-09:
In the Documentation, I see we have the

```
Test/runMain vexiiriscv.tester.TestBench --with-mul --with-div --load-elf ext/NaxSoftware/baremetal/dhrystone/build/rv32ima/dhrystone.elf n --trace-all
```

to run a test on the executable `dhrystone.elf` which is a self contained program.

My use case I have applications (program) like, FFT, Dijkstra, Qsort, SHA which once compiled, need to read data from a file usually provided a a command line argument

```
./fft 8 32768 > output.txt
./dijkstra_large input.dat > output_large.txt
./sha input_large.asc > output_large.txt
```

I am new to Scala, SBT and SpinalHDL. Is there a work around for this in the framework? Any Support from VexiiRiscv?

Dolu1990, 2025-04-13:
Hi,

It isn't a testbench feature. but a C/C++ RISC-V software one.
The testbench already allows you to load binary files into the memory.
What you would need is some C/C++ software which is capable to read/write a file system which is in the RAM.

nishanthsmurthy24, 2025-04-16:
I managed to convert a data file into a header that the Program reads.

I used the flow described in [Self Contained Tutorial - VexiiRiscv Documentation](https://spinalhdl.github.io/VexiiRiscv-RTD/master/VexiiRiscv/Tutorial/index.html#c-code-hello-world-literally)

I went through the recursive Makefile calls and then put together the important and required part into this simplified Makefile

```
# Source files
# FILE1 = src/dijkstra_small.c
FILE2 = src/dijkstra_large.c
STDLIB = ../../baremetal/common/stdlib.c
STRCMP = ../../baremetal/common/strcmp.S

STARTUP = ../../baremetal/common/start.S
LINKER_SCRIPT = ../../baremetal/common/app.ld

# Toolchain and flags
CC = riscv32-unknown-elf-gcc
# LD = riscv32-unknown-elf-ld
CFLAGS = -c -O2 -ffunction-sections -march=rv32gc_zicsr -mabi=ilp32 -mcmodel=medany
CFLAGS += -fno-common -fno-inline -falign-functions=8 -falign-jumps=8 -falign-loops=8 -mtune=sifive-7-series
CFLAGS += -Wno-implicit-int -Wno-implicit-function-declaration
CFLAGS += -DTIME -DCORE_HZ=12000000ll
CFLAGS += -DUSE_GP
CFLAGS += -I../../baremetal/driver

ASFLAGS = $(CFLAGS)
LDFLAGS = -lc -lgcc -nostartfiles -ffreestanding -Wl,-Bstatic -T $(LINKER_SCRIPT)
LDFLAGS += -L../../baremetal/common

# Output
BUILD_DIR = build

# Targets
all: $(BUILD_DIR)/dijkstra_large.elf

# Ensure build directory exists
$(BUILD_DIR):
	mkdir -p $(BUILD_DIR)

# Compile custom stdlib.c and strcmp.S
$(BUILD_DIR)/stdlib.o: $(STDLIB) | $(BUILD_DIR)
	$(CC) $(CFLAGS) -o $@ $<

$(BUILD_DIR)/strcmp.o: $(STRCMP) | $(BUILD_DIR)
	$(CC) $(ASFLAGS) -o $@ $<

# Compile start.S
$(BUILD_DIR)/start.o: $(STARTUP) | $(BUILD_DIR)
	$(CC) $(ASFLAGS) -o $@ $<

# Compile C files
# $(BUILD_DIR)/dijkstra_small.o: $(FILE1) | $(BUILD_DIR)
# 	$(CC) $(CFLAGS) -o $@ $<

$(BUILD_DIR)/dijkstra_large.o: $(FILE2) | $(BUILD_DIR)
	$(CC) $(CFLAGS) -o $@ $<

# Link with riscv32-unknown-elf-ld
# $(BUILD_DIR)/dijkstra_small.elf: $(BUILD_DIR)/start.o $(BUILD_DIR)/dijkstra_small.o $(BUILD_DIR)/stdlib.o $(BUILD_DIR)/strcmp.o
# 	$(CC) $(LDFLAGS) -o $@ $^

$(BUILD_DIR)/dijkstra_large.elf: $(BUILD_DIR)/start.o $(BUILD_DIR)/dijkstra_large.o $(BUILD_DIR)/stdlib.o $(BUILD_DIR)/strcmp.o
	$(CC) $(LDFLAGS) -o $@ $^

# Clean
clean:
	rm -rf $(BUILD_DIR)
```
The compilation works well as expected, no issues there. But the simulation does not go through. Here is the snapshot of the Sim Run

```
[info] ### Stats ###
[info] kind :  miss / times     miss  taken
[info] J/B  :     0 /     1   0.0%   0.0%
[info]   B  :     0 /     1   0.0%   0.0%
[info] Dispatch  0   :   14895 /   16076  92.6%
[info] Dispatch  1   :    1180 /   16076   7.3%
[info] Candidate 0   :   14895 /   16076  92.6%
[info] Candidate 1   :    1180 /   16076   7.3%
[info] Dispatch halt :      24 /   16076   0.1%
[info] Execute  halt :       0 /   16076   0.0%
[info] IPC           :      12 /   16076   0.0%
[error] Exception in thread "main" spinal.sim.SimFailure: Vexii hasn't commited anything for too long, last uop id 0x484
```

I compared the log files for the simulation of `ext/NaxSoftware/baremetal/dhrystone_vexii/build/rv32imafdc/dhrystone_vexii.elf` and the ELF i generated from the above Makefile.

Any reason why this is happening? How can I get the simulation to run? Attached are the Objdumps and tracer logs.

[dijkstra_tracer.log](https://github.com/user-attachments/files/19780486/dijkstra_tracer.log)

[dhrystone_tracer.log](https://github.com/user-attachments/files/19780491/dhrystone_tracer.log)

[dijkstra_large_objdump.log](https://github.com/user-attachments/files/19780527/dijkstra_large_objdump.log)

[dhrystone_vexii_objdump.log](https://github.com/user-attachments/files/19780528/dhrystone_vexii_objdump.log)

Dolu1990, 2025-04-17:
rv commit 0 000000008000002c
rv store bro 0 11
rv trap 0 0 2

Mean PC 0x8000002C failed with a exception code 2  => illegal instruction
Check what is at 0x8000002C, that will tell you which instruction wasn't enabled in the VexiiRiscv config you used.

nishanthsmurthy24, 2025-05-07:
> rv commit 0 000000008000002c rv store bro 0 11 rv trap 0 0 2
> 
> Mean PC 0x80000030 failed with a exception code 2 => illegal instruction Check what is at 0x80000030, that will tell you which instruction wasn't enabled in the VexiiRiscv config you used.

Can you please list all the trap codes that can be caught in the tracer.log? I searched it in the documentation and the source codes, but is not very clear.

Dolu1990, 2025-05-07:
trap codes refer to the RISC-V mcause ones. Check the risc-v privileged spec for the mcause exception codes.

nishanthsmurthy24, 2025-05-08:
@Dolu1990 I checked the riscv-privileged spec and I have 

```
rv store com 0 2988
rv commit 0 0000000080000048
rv store bro 0 2988
rv rf w 0 0 32 ffffffff800100d8
rv commit 0 000000008000004c
rv commit 0 0000000080000050
rv trap 0 0 0
rv trap 0 0 1
rv trap 0 0 1
rv trap 0 0 1
rv trap 0 0 1
```

These traps correspond to 

[image]

This trap is a part of the init part before the main()

The compilation log is as follows
```
riscv32-unknown-elf-gcc -c -O2 -ffunction-sections -march=rv32i_zicsr -mabi=ilp32 -mcmodel=medany -fno-common -fno-inline -falign-functions=8 -falign-jumps=8 -falign-loops=8  -Wno-implicit-int -Wno-implicit-function-declaration -DTIME -DCORE_HZ=12000000ll -mtune=sifive-7-series -DUSE_GP -I../../baremetal/driver -o build/start.o ../../baremetal/common/start.S
riscv32-unknown-elf-gcc -c -O2 -ffunction-sections -march=rv32i_zicsr -mabi=ilp32 -mcmodel=medany -fno-common -fno-inline -falign-functions=8 -falign-jumps=8 -falign-loops=8  -Wno-implicit-int -Wno-implicit-function-declaration -DTIME -DCORE_HZ=12000000ll -mtune=sifive-7-series -DUSE_GP -I../../baremetal/driver -o build/fft_large.o src/fft_large.c
riscv32-unknown-elf-gcc -c -O2 -ffunction-sections -march=rv32i_zicsr -mabi=ilp32 -mcmodel=medany -fno-common -fno-inline -falign-functions=8 -falign-jumps=8 -falign-loops=8  -Wno-implicit-int -Wno-implicit-function-declaration -DTIME -DCORE_HZ=12000000ll -mtune=sifive-7-series -DUSE_GP -I../../baremetal/driver -o build/fftmisc.o src/fftmisc.c
riscv32-unknown-elf-gcc -c -O2 -ffunction-sections -march=rv32i_zicsr -mabi=ilp32 -mcmodel=medany -fno-common -fno-inline -falign-functions=8 -falign-jumps=8 -falign-loops=8  -Wno-implicit-int -Wno-implicit-function-declaration -DTIME -DCORE_HZ=12000000ll -mtune=sifive-7-series -DUSE_GP -I../../baremetal/driver -o build/fourierf.o src/fourierf.c
riscv32-unknown-elf-gcc build/start.o build/fft_large.o build/fftmisc.o build/fourierf.o -o build/fft_large.elf -lm -lgcc -nostartfiles -ffreestanding -Wl,-Bstatic -T ../../baremetal/common/app.ld -L../../baremetal/common
/project/sartori00/riscv/riscv32/lib/gcc/riscv32-unknown-elf/14.2.0/../../../../riscv32-unknown-elf/bin/ld: warning: build/fft_large.elf has a LOAD segment with RWX permissions
```

I don't how the instruction address is being misaligned. Is it something to do with the Compiler ? or a bug in the design?

[fft_large_tracer.log](https://github.com/user-attachments/files/20105953/fft_large_tracer.log)

Dolu1990, 2025-05-08:
would need to see your software .asm

nishanthsmurthy24, 2025-05-08:
I used the default linker script `app.ld` and `start.S` provided in 

`ext/NaxSoftware/baremetal/common`

and _none_ of `stdlib.c` and `strcmp.S`

I also see that there is a `trap.S` in the directory mentioned above. I am _not_ using that. Is that helpful?

[src.zip](https://github.com/user-attachments/files/20110203/src.zip)

nishanthsmurthy24, 2025-05-09:
Attached ELF file 
[fft_large.elf.zip](https://github.com/user-attachments/files/20130390/fft_large.elf.zip)

nishanthsmurthy24, 2025-06-02:
Any updates on this @Dolu1990 ? Thanks in advance.

closed 2025-06-02 by nishanthsmurthy24