SpinalHDL/VexiiRiscv #54: Potential store-conditional return value bug in `LsuCachelessPlugin`

mrcmry, 2025-04-26:
I do not understand the following code in `LsuCachelessPlugin.scala`:

```
      iwb.valid := SEL && !FLOAT
      iwb.payload := rspShifted.resized

      if (withAmo) when(ATOMIC && !LOAD) {
        iwb.payload(0) := onJoin.SC_MISS
        iwb.payload(7 downto 1) := 0
      }
```

Why is it not `iwb.payload(31 downto 1) := 0` or simply only setting bit 0 ? Is it something with the shifter before that (but the shifter should go away if `LSLEN` == `XLEN`) ? If the data response from the memory is always 0 on a write, why setting bit 7 to 1 to 0 ?

[The doc for `SC.W`](https://msyksphinz-self.github.io/riscv-isadoc/html/rva.html#sc-w) states "SC writes zero to rd on success or a nonzero code on failure".

Dolu1990, 2025-05-01:
Ahhh that one is a tricky one : 
https://github.com/SpinalHDL/VexiiRiscv/blob/dd2fb21cab62d68408fc402e6a9557f2d61f9a0f/src/main/scala/vexiiriscv/execute/lsu/LsuCachelessPlugin.scala#L98
=>
https://github.com/SpinalHDL/VexiiRiscv/blob/dd2fb21cab62d68408fc402e6a9557f2d61f9a0f/src/main/scala/vexiiriscv/riscv/Rvi.scala#L178

In other words, Vexiiriscv handle store conditional instruction by zero extenting them for everything above bit 7.
It is done like that to save area by reusing the zero/sign extention hardware used by other instructions ^^

mrcmry, 2025-05-03:
Ahaa, now I see ! Thanks for the response.

Very clever. I will add a comment about that in my next spelling/typo PR.

closed 2025-05-03 by mrcmry