SpinalHDL/VexiiRiscv #57: [Feature Request] Vector extensions

jrmoserbaltimore, 2025-05-24:
I was thinking on how to do this and it turned out very complicated, so I figured I would put my thoughts here on vector extensions.

# Vector ops are heavy

First, vector extensions are heavy.  This is why I'm only thinking of the Zve* vector ops:  there's going to be a lot of important use cases for not bringing all the vector ops, only the vector ops you need.

The lightest example I can come up with is **Zve32x** (integer only, 32 bit) with VLEN=1, which requires supporting SEW of 8, 16, and 32, only on one hart of a multi-hart system, while stalling dual-issue.  That allocates an additional 32 32-bit registers, which by LMUL can be grouped in up to 8 32-bit, 16 16-bit, or 32 8-bit elements.  With hard multipliers that allow 1 32x32, 4 16x16, or 8 8x8 multipliers (for example, ECP5 sysDSP 36x36, 18x18, or 9x9) requires 10 DSP blocks:

- 4 8x8 blocks
  - 32 8x8
  - 8 16x16
  - 2 32x32
- 2 16x16 blocks
  - 8 16x16
  - 2 32x32
- 4 32x32 blocks

That requires some glue to combine smaller multiplies into bigger multiplies, where the result of 4 8x8 multiplications can be shifted and added in such a way to produce a 16x16 multiply, and the result of 4 16x16 multiplications can be shifted and added in such a way to produce a 32x32 multiply.  Using the 8x8 blocks then needs 7 layers of carry-save followed by a carry-propagate add:

- 32 partial sums become 15+2
- These become 10+2
- These become 6
- These become 4, then 3, then 2
- Use a fast adder on the remaining 2

Or else, one could use 4 8x8, 4 16x16, and 8 32x32 (16 DSP blocks, or 32 of the available 18x18 multipliers) and Mux between them to trade off DSP block usage for LUT usage and get a shorter pipeline or higher fmax (this is probably the better option).  On DSP48E you only get 25x18 multipliers, so you need to implement all 32-bit operations as sets of 4 DSPs, meaning the smallest possible DSP usage is 32 for 8 32-bit operations.  The multipliers don't subdivide, so the whole 18x32 multiplier is used to produce an 8x8 as well.

That's just to support 32-bit integer vectorizations on one hart with single-issue, making the dual-issue dispatcher just stall if it sees two vector ops.

So yeah.  That's a bunch of stuff just to get the vector ops in but limit the amount of resources consumed, and we're still talking about consuming 32 18x18 multipliers.

# Vector ops require caching adjustments

If you want fast loads and stores on vector ops, you need to access half a cache line in the pipeline:  when you read the D$ cache, you read a 256-bit wide data bus so you can fill a vector register or copy it back to a cache line.  It is totally legal for a RISCV processor to throw an alignment error if you attempt to load/store a vector group misaligned with the group size, e.g. if LMUL=8 then you must align to vector load/store to 32 bytes, if LMUL=4 then must align to 16 bytes, if LMUL=2 then must align to 8 bytes, and if LMUL=1 then must align to 4 bytes.  This approach simplifies loading and storing by preventing breaks across the bus width.

# Dual-issue considerations

Even if you can't dual-issue vector operations, you can dual-issue ops and load/store.  For example, you can't dual-issue multiplies, but you can issue a vector multiply and a vector add using separate sets of vector registers (if you don't have a hazard) because you have adders and multipliers separate; or you can issue a load/store along with math.  A dual-issue implementation doesn't need to allow dual-issue of everything.

# Multi-hart considerations

All of the above for Zve32x gets duplicated to other harts if you allow multiple harts to have Zve32x extensions.  Likewise, if you also enable floats, you need to deal with that with the FPU.

For this reason, it's useful to be able to control which harts have vector extensions and which don't.  I actually have an application where a RTOS kernel provides a driver for a GPU, which becomes MUCH simpler and consumes thousands fewer LUT if I precalculate several derivatives, square roots, inverses, logarithms, and exponents.  That consumed 25% of a core just for derivatives and inverses, but with vector ops it would drop to 3%.  I could squeeze that further by extending with custom vector ops instructions, which can do one-cycle log2 and exp2 to low but adequate precision for my needs (two rounds of Newton-Rhapson turn this into high precision), which lets me do a three cycle rough square root.

All of that sounds insane when you think about how you have to add Zve32x and custom instructions anyway and where's the savings now, but dropping the CPU usage opens that hart up to do things like physics and AI calculations that can be vectorized, which suddenly becomes very meaningful.

That's a real application boost:  instead of using around 3,000 extra LUT in a peripheral (which is actually too slow, so I use 3 cores of that, so 9,000 total LUT consumed) I offloaded to CPU, but that ate a ton of CPU; implementing vector extensions turns this into a trivial 2%-3% of CPU usage; and the vector extensions allow other very heavy workloads that heavily benefit from vectorizations to be done on that hart at similarly trivialized cost.  Big win.

# Recap

So to recap:

- Vector ops bring more registers, more ALU resources, and more complexity
- Zve* vector ops let you pick subsets of vector ops and infrastructure to add, which should be configurable
- We need a wide data bus to D$ for load/store if we want to avoid 2.7a bottleneck, especially in tight load-operate-store loops
- Requiring alignment to VLEN×LMUL is valid and makes things simpler
- Dual-issue can work if you just don't issue the same kind of instruction, for example a multiplication and an addition is fine but two multiplications isn't unless you invest a ton more resources
- You can have vector extensions without having them on all harts, useful in specialized embedded systems
- There are real-world applications where vector ops are make-or-break by an enormous margin for feasibility

And, of course

- Implementing vector ops is a huge lift and nobody should expect this to happen by next week :)

Just my thoughts.

jrmoserbaltimore, 2025-05-29:
No kidding! I suggested using the integer registers once and was ridiculed for the idea years back because apparently it's a bad idea to reuse registers for different things, meanwhile they were already working on Zfinx.  I'd use single-width registers and set LMUL to 2, 4, or 8 if I was building a CPU with Zve32x, so the vector registers would be the same size as the integer registers, so it doesn't make sense to me that they don't have like Zvinx—especially since that would let you mass-load the integer registers and then do scalar ops, which would be cool!

franktaTian, 2026-03-04:
I think  Zve32x is needed for VexiiRiscv. Ref.  google's coralnpu.