SpinalHDL/VexiiRiscv #58: How to tunnel VexiiRiscv Debug port using BSCANE2?

Matze584, 2025-05-27:
Hello!

I am trying to establish a debug link to my VexiiRiscv-SOC. 
I would be very happy, if the JTAG interface to Vexii could be tunneled through BSCANE2 like in VexRiscv. 
There's simply not enough free I/O on my board, to patch a second JTAG Port onto it.

The Docs say, its possible to tunnel JTAG through the FPGAs native JTAG as before on Vex.
But i can't find any details on how to generate Vexii with BSCANE2 tunneling.
What would be needed to do this ?

Thank you much!! :-)

Dolu1990, 2025-05-27:
Hi,

What SoC are you using ? i mean, do you use the microsoc template, or something like it ? or you have your own verilog/vhdl soc and use vexiiriscv as a verilog file ?

Dolu1990, 2025-05-27:
With the commit above, you can now generate the CPU with it.
You need --debug-privileged --debug-jtag-instruction arguments.

=> 
```verilog
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_tdi,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_enable,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_capture,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_shift,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_update,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_reset,
  output wire          EmbeddedRiscvJtag_logic_jtagInstruction_tdo,
  output wire          EmbeddedRiscvJtag_logic_ndmreset,
  input  wire          reset,
  input  wire          clk,
  input  wire          EmbeddedRiscvJtag_logic_debug_reset,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_tck
```

So, for the reset, you need kinda a external reset controller.

Here is the reset herarchy / order in which they should be released :
toplevel power on reset / external reset  -> global reset -> EmbeddedRiscvJtag_logic_debug_reset  -> reset

EmbeddedRiscvJtag_logic_jtagInstruction_tck is only used on 
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_tdi,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_enable,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_capture,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_shift,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_update,
  input  wire          EmbeddedRiscvJtag_logic_jtagInstruction_reset,
  output wire          EmbeddedRiscvJtag_logic_jtagInstruction_tdo,

For the openocd tcl script, this configuration should be fine : 
https://github.com/SpinalHDL/VexiiRiscv/blob/dev/src/main/tcl/openocd/vexiiriscv_jtag_tunneled.tcl
(so riscv use_bscan_tunnel 6 1) But i'm not sure which bscane id was used. Seems like it was the one mapped on jtag instruction ID 0x23

Matze584, 2025-05-27:
That's amazing! Thanks a lot! 
It's working beautifully :-)

The JTAG ID for the BSCANE2 needs to be 4, which corresponds to Jtag Instruction ID 0x23 in the provided 
OpenOCD Script. 

Just one more question: What is ndmreset meant to do? 

BTW: is there a way to donate for your work and effort with Vex(ii)Riscv and its ecosystem?

Dolu1990, 2025-05-28:
> It's working beautifully :-)

Nice :D I was expecting it to be more painfull ^^

> Just one more question: What is ndmreset meant to do?

It allows openocd to reset the SoC side of things (but not the debug side)

> BTW: is there a way to donate for your work and effort with Vex(ii)Riscv and its ecosystem?

Thanks :)

- For VexiiRiscv, the more inline channel would be : https://github.com/sponsors/Dolu1990 , as it is to cover living expenses
- For the ecosystem (the SpinalHDL language), there is https://opencollective.com/spinalhdl . this is more to cover traveling expenses when going in hardware events.