SpinalHDL/VexiiRiscv #64: Support for Regfile with fewer than 32 registers

Jzjerry, 2025-07-18:
Hi,

I've been working on some RVV-like custom instructions, in which I added a vector register file. 

It seems all regfile addresses are defaultly 5 bits in Vexii, but for vector regfile, it would be quite huge to have 32 registers with long data widths (like >128 bits). So I tried to build a register file with fewer registers (like 8 or 16), but there are some width mismatches for writeback and bypasser.

I find they are pretty much related to `physWidth` and `PHYS` in DecoderPlugin, so it should be possible to support different reg nums by making these two variables infer the regfile address width at generation time🤔.

Considering there might be a need for sth like RV32E/RV64E, it would be nice to support regfiles with <32 registers👀.

Jzjerry, 2025-07-18:
I tried to make a simple adaptation for RV32E (c539f0dc097bd3225ab1992dec61972c5d094712), it basically works, passing generation and coremark simulation, but still does not work for the case if I want a 16x IntRegFile and a 32x FloatRegFile.

Seems like I need to add some more decoding logic here...

Dolu1990, 2025-07-19:
> Btw, I think the AES option is on by default for now, so xlen=64 is asserted XD.

Hooo thanks, that was a mistake, i pushed a fix.

I will be back from holiday / trip the 26 ^^

Dolu1990, 2025-07-28:
Merged, thanks ^^
Remains floating point support, but lets see if somebody realy want it, given RVE isn't realy for floating point cpus

closed 2025-07-28 by Jzjerry

Dolu1990, 2025-07-28:
> Yeah, rv32e actually conflicts with f in riscv gcc.

Ahhhh good to know XD
Thanks ^^