SpinalHDL/VexiiRiscv #71: MicroSoC generation failing with CFU plugin

Rishu31, 2025-08-20:
Hi, I tried to generate a MicroSoC version with CFU support with the following command

`sbt "runMain vexiiriscv.soc.micro.MicroSocGen --with-mul --with-div --with-cfu"`

Got error message saying this for `cmd_ready, rsp_valid, and rsp_output_0` variables
`NO DRIVER ON (toplevel/system_cpu_logic_core/CfuPlugin_logic_bus_cmd_ready : in Bool) (combinatorial signal with no logic connection), defined a
[error]     at spinal.lib.Stream.<init>(Stream.scala:99)
[error]     at spinal.lib.StreamFactory.apply(Stream.scala:60)
[error]     at spinal.lib.StreamFactory.apply(Stream.scala:65)
[error]     at vexiiriscv.execute.cfu.CfuBus.<init>(CfuBus.scala:63)
[error]     at vexiiriscv.execute.cfu.CfuPlugin$$anon$1.<init>(CfuPlugin.scala:71)
[error]     at vexiiriscv.execute.cfu.CfuPlugin.$anonfun$logic$1(CfuPlugin.scala:63)
[error]     at spinal.lib.misc.plugin.FiberPlugin$$anon$1.$anonfun$setup$4(Fiber.scala:71)
[error]     at spinal.lib.misc.plugin.PluginHost.rework(Host.scala:29)
[error]     at spinal.lib.misc.plugin.FiberPlugin$$anon$1.$anonfun$setup$2(Fiber.scala:69)
[error]     at spinal.lib.misc.plugin.FiberPlugin$$anon$1.$anonfun$setup$1(Fiber.scala:66)`

Also, how to disable Privileged mode in MicroSoC? The documentation does not touch upon these things in much detail.

Could you please help with these things?

Thanks

Jzjerry, 2025-08-21:
I'm also trying to connect a CFU to vexii in MicroSoc.

> MicroSoc do not connect the CFU interface to anything

You can try connecting a CFU to the Vexii's CFU interface by adding the following in `TilelinkVexiiRiscvFiber` around Line 103:
```scala
import vexiiriscv.execute.cfu.{CfuPlugin, CfuBus, CfuTest}
class TilelinkVexiiRiscvFiber(plugins : ArrayBuffer[Hostable]) extends Area with RiscvHart{

// ......

    //Connect stuff
    plugins.foreach {
      case p: PrivilegedPlugin => {
        val hart = p.logic.harts(0)
        hart.int.m.timer := priv.get.mti.flag
        hart.int.m.software := priv.get.msi.flag
        hart.int.m.external := priv.get.mei.flag
        priv.get.stoptime := p.p.withDebug.mux(hart.debug.stoptime, False)
        if (p.p.withSupervisor) hart.int.s.external := priv.get.sei.flag
        if (p.p.withRdTime) p.logic.rdtime := priv.get.rdtime
      }
      case p: CfuPlugin => {
        val bus = p.logic.bus
        val test = CfuTest()
        test.io.bus << bus
      }
      case _ =>
    }
```

It should allow you to generate a MicroSoc with CFUs, just replace `CfuTest()` with your own CFU and make sure the bus params match. 

Just a demo, I haven't done any simulation yet XD.

Rishu31, 2025-11-03:
Hi, I have incorporated all of the changes suggested, and now a setup with a combinational CFU works seamlessly.
However, CFU instructions which take e.g. 3 cycles to be completed and resp_valid is being delayed accordingly, the output is typically a garbage value (I have observed different outputs every run, despite no changes in the code).

Help appreciated.

Rishu31, 2025-11-04:
Hey
I did a few more experiments; it seems to work properly when I reduce the operating frequency, though the bitstream generation timing had succeeded before as well.
Did not expect that to be the problem.
Sorry for the trouble :)