SpinalHDL/VexiiRiscv #80: Trying to connect a LsuL1Cmd port from external units (like CFU) to CPU

Jzjerry, 2025-09-03:
Hi,

Still messing with CFU and SoC stuff.

I've successfully built a CFU that connects with both the CPU and the system bus (without a cache). 
It works well, but meets some coherence problems when the CPU gets an L1, so I basically have to put a fence before executing CFU instructions, which leads to some performance drop.

So I wonder if I could directly share the L1 path for CFU and CPU, and find out that I can add a `LsuL1Cmd` bus to `LsuPlugin`. But when I try to build a CFU with a `Stream(LsuL1Cmd)`, it leads to a `ScopeProperty spinal.lib.misc.database.Database$@2ae2ab2 isn't set`. I believe that's because some globals are not set yet in Vexii, but CFU requires them. The code for my CFU fiber is like:

```scala
class CfuFiber() extends Area {

  val logic = Fiber build new Area {

    val cfuParam = TilelinkCfuFiber.getCfuBusParameters
    val cfuBus = CfuBus(cfuParam)

    val l1Bus = Stream(LsuL1Cmd())
    val cfu = new DemoCfuL1(cfuParam)

    cfu.io.bus <> cfuBus
    cfu.io.l1 <> l1Bus
  }
}
```

I'm a bit confused about the elaboration schedule in Vexii. How should I make the fiber wait utill the LsuL1Cmd is ready to be used?
And also maybe sharing an L1 is not a good idea at all XD?

Dolu1990, 2025-09-05:
> So I wonder if I could directly share the L1 path for CFU and CPU,

Currently this isn't realy supported, but there may be ways to reuse the feature that the MMU uses to read the L1 cache to walk pages. See https://github.com/SpinalHDL/VexiiRiscv/blob/989c64796317552c33337ebc2749592c6cbe55cc/src/main/scala/vexiiriscv/memory/MmuPlugin.scala#L182
Just that this feature was never tested for memory store, and never tested when the CPU is already busy with instructions

LsuL1Cmd is more an internal thing of the LsuPlugin, there is no real spot to plug it from the outside, better use the things like the MMU does.

Dolu1990, 2025-09-07:
> For now, the CFU doesn't respond until its mem access ends, so I think it's still kinda coupled? But yeah, there could be cases where access is decoupled in the future.

Ahhh note that if the CPU pipeline is halted by an instruction the accessBus will not progress either.

>  lsu-l1-coherency

It require to have either the HubFiber or the CacheFiber to connect all the coherent agents.

Also, coherency won't prevent the needs for fence.

Jzjerry, 2025-09-07:
> the accessBus will not progress either
> coherency won't prevent the needs for fence.

Oops. Then it looks like I have to keep the `fence` for now. It's an approximate 5% more cycles for my workload.